rust-embedded/svd2rust #205: Why is the API so verbose?

bergus, 2018-05-02:
I'm not sure whether this is a question about the library design or a feature request for simplifying the read/modify/write API.

Why do we need to write
```rs
peripheral.register.write(|w| w.field().set_bit())
peripheral.register.write(|w| w.field().clear_bit())
peripheral.register.write(|w| w.field().bit(bool_val))
```
instead of
```rs
peripheral.register.write(|w| w.set_field())
peripheral.register.write(|w| w.clear_field())
peripheral.register.write(|w| w.field(bool_val))
```
?

All those field-R and field-W types seem to considerably blow up the API (and [file size](https://github.com/japaric/svd2rust/issues/116)). The mentioning of "bit" or "bits" in the method call seems to be superfluous clutter, not making the API more fluent. I know the type of the register fields that I work with, or I can infer it from passing either a boolean or a number to the method.

What is the advantage of these additional types?

(Regarding changing or extending the API, I have not enough experience with the ecosystem to judge how to best proceed with this in terms of backwards compatibility and at the same time not bloating the API further).

----

Also there are quite a few registers that hold only a single field anyway. Would it be possible to allow syntax like
```
peripheral.register.read_field()
peripheral.register.write_field(num)
peripheral.register.modify(|val| val)
```
for them instead of having to deal with `.bits()` everywhere?

Or am I mistaken about the scope of svd2rust and it is the job of the HAL crates to provide these abstractions?

Emilgardis, 2018-05-02:
HAL is used to abstract over the different resources of the chip and make ease of use better. The api `svd2rust` generates is used as a "low-level" interface. To answer your question it's correct that HAL is supposed to provide abstractions, but not in that exact fashion of shortening field access, but to handle resources in a more convenient way.

Regarding why we use `R` and `W`, or specifically why we use

```rust 
peripheral.register.write(|w| w.field().set_bit())
``` 

and not 

```rust
peripheral.register.write(|w| w.set_field())
```

Consider this call and how it would be used with the second variant

```rust
peripheral.register.write(|w| w.field().open())
```

To transform enumerated values we'd have to do

```rust
peripheral.register.write(|w| w.set_field_as_open())
```
Which to me doesn't seem like a good idea.

-------

Eitherway I do agree with you that it is a bit superfluous, but I think that it is fine because the api generated is supposed to be as low-level as possible/convenient.

bergus, 2018-05-02:
Ah, thanks for the insight. I have not yet worked with enumerated values yet, I guess the API is more fluent with those. I would have thought of
```rs
use per::reg::Field::*
peripheral.register.write(|w| w.field(Open))
```
only, which indeed isn't as nice as the convenience methods.

Emilgardis, 2018-05-02:
No problem! 👍 

Feel free to close the issue if you feel like it's been resolved.

closed 2018-05-06 by Emilgardis