rust-embedded/svd2rust #218: Use `AsMut`/`AsRef`-like deref instead of unions

Emilgardis, 2018-05-14:
Currently, `svd2rust` supports overlapping registers via the cli flag `--nightly`. This will generate an `union` of all registers that overlap each other.

Since it is not expected that non-`Copy` fields in unions will be stabilized soon (see tracking issue rust-lang/rust#32836), another system for access would prove beneficial for library maintainers and users wanting to access overlapping registers using stable rust.

cc @wez @japaric

wez, 2018-05-15:
Yeah, makes sense.  Referencing this example from atsamd21:

https://docs.rs/atsamd21g18a/0.2.1/atsamd21g18a/tc3/union.RegisterBlock.html

instead of the union generate something like this:

```
pub struct RegisterBlock {}

impl RegisterBlock {
    fn as_count8_mut(&mut self) -> &mut COUNT8 {
        unsafe {
            &mut *((self as *mut Self) as *mut COUNT8)
        }
    }

    fn as_count8_ref(&self) -> &COUNT8 {
        unsafe {
            &*((self as *const Self) as *const COUNT8)
        }
    }

    fn as_count16_mut(&mut self) -> &mut COUNT16 {
        unsafe {
            &mut *((self as *mut Self) as *mut COUNT16)
        }
    }

   // as_count16_ref(), as_count32_mut(), as_count32_ref() here
}
```

I've found that in `atsamd21-hal` I'd often make a safe accessor for the union anyway, so this removes some boilerplate for me.

wez, 2018-05-15:
I can play around with this, since I think I'm the only person currently with a crate that requires and is using this :)

wez, 2019-01-04:
I have some WIP in https://github.com/wez/svd2rust/tree/no_unions
It happens to look like the suggestion in https://github.com/rust-embedded/svd2rust/issues/230

adamgreig, 2019-07-26:
I believe this is now solved by #318.

closed 2019-07-26 by adamgreig