rust-embedded/svd2rust #219: v0.13.0 as-installed emits interrupt! with STATE, which doesn't match repo

thenewwazoo, 2018-05-14:
I've got svd2rust 0.13.0 installed:
```
> svd2rust -h
svd2rust 0.13.0 ( )
Generate a Rust API from SVD files

USAGE:
    svd2rust [FLAGS] [OPTIONS]
...
```

It emits the following `interrupt` macro:
```rust
macro_rules! interrupt {
    ($Name:ident, $handler:path,state: $State:ty = $initial_state:expr) => {
        #[allow(unsafe_code)]
        #[deny(private_no_mangle_fns)]
        #[no_mangle]
        pub unsafe extern "C" fn $Name() {
            static mut STATE: $State = $initial_state;
            let _ = $crate::Interrupt::$Name;
            let f: fn(&mut $State) = $handler;
            f(&mut STATE)
        }
...
```

...but the [docs](https://docs.rs/svd2rust/0.13.0/svd2rust/macro.interrupt.html) and the [repo](https://github.com/japaric/svd2rust/blob/a8ee2d98c1d2aa3404d520546564b7e752f76d56/src/lib.rs#L476) refer to `locals`.

Emilgardis, 2018-05-15:
I'm not sure I understand what this issue is for, the macro you are showing is in [interrupt.rs#L126](https://github.com/japaric/svd2rust/blob/v0.13.0/src/generate/interrupt.rs#L126) 
The exported `interrupt!` macro is an artifact from earlier versions, since `svd2rust` generates it for you now depending on target.

japaric, 2018-05-15:
@thenewwazoo 

> but the docs 

says that the interrupt! macro looks different on Cortex-M. Quoting the docs

> NOTE The interrupt! macro on Cortex-M device crates is closer in syntax to the exception! macro. This documentation doesn't apply to it. For the exact syntax of this macro check the documentation of the device crate you are using.

And the device crate properly document the macro. See the [stm32f30x] for example.

[stm32f30x]: https://docs.rs/stm32f30x/0.7.0/stm32f30x/macro.interrupt.html

thenewwazoo, 2018-05-15:
Oh, I see now. I found the "closer in syntax" note to be confusing - the distinction here is between the (apparently historical?) macro in the `svd2rust` crate itself and the macro it _generates_. I might reword that as:
> NOTE This documentation is for a macro exported _only_ from the `svd2rust` crate. The `interrupt!` macro generated by svd2rust for Cortex-M device crates is different from this macro, and mirrors the `exception!` macro. For that macro, see the documentation for the generated device crate.

closed 2018-05-15 by thenewwazoo