rust-embedded/svd2rust #222: Adding --nightly flag breaks register padding

pftbest, 2018-05-29:
When I compare the output of svd2rust with `nightly` flag and without it, I see that the padding is incorrect

```
$ svd2rust -i msp430g2553.svd --target=msp430 | rustfmt > stable.rs
$ svd2rust --nightly -i msp430g2553.svd --target=msp430 | rustfmt > nightly.rs
```

```diff
@@ -23885,7 +23886,6 @@
     #[doc = r" Register block"]
     #[repr(C)]
     pub struct RegisterBlock {
-        _reserved0: [u8; 1usize],
         #[doc = "0x01 - Basic Clock System Control 3"]
         pub bcsctl3: BCSCTL3,
         _reserved1: [u8; 2usize],
@@ -26634,7 +26634,6 @@
     #[doc = r" Register block"]
     #[repr(C)]
     pub struct RegisterBlock {
-        _reserved0: [u8; 1usize],
         #[doc = "0x01 - Comparator A Control 1"]
         pub cactl1: CACTL1,
         #[doc = "0x02 - Comparator A Control 2"]
@@ -28928,7 +28927,6 @@
```

Emilgardis, 2018-05-29:
I think this is due to [#L470](https://github.com/japaric/svd2rust/blob/8a50bc153e14c907695e920c449f9efcf074ed3d/src/generate/peripheral.rs#L470). Does the output show a warning for those registers?

cc @wez, you are most familiar with the `--nightly` logic, could you explain how to implement this?

pftbest, 2018-05-29:
No, there were no warnings, here is the svd file if you want to try it on your side:

[msp430g2553.svd.txt](https://github.com/japaric/svd2rust/files/2048105/msp430g2553.svd.txt)

commit d5e5b4de (2018-08-07): Merge pull request #223 from Emilgardis/fix-nopad

Fixes #222, no padding emitted when pad is first in block
commit d86365bf (2018-08-07): Fixes #222, no padding emitted when pad is first in block