rust-embedded/svd2rust #225: derivedFrom and missing modules ... is there a solution I am missing?

baldwin1974, 2018-06-10:
Apologies if there some place where this is addressed and I have missed it. In searching the issues here, I've seen that there are allusions to the derivedFrom attribute and the problems it causes, but I am not sure if there is a solution to it. In the STM32H7x3 svd file, there are numerous peripherals that derived from a first and single instance, and svd2rust has created a crate for me (stm32h743) that lacks those "derivedFrom" modules (e.g., gpioa exists, but gpiob through gpiok). Yet, even though there aren't the modules, there are plenty of appropriate structs (e.g., GPIOA through GPIOK).  Do I simply implement code through those structs, and if so, can someone point me to an example? 

Or should I simply go through the original svd file and replace the "derivedFrom" section with similar code from the first instances, then re-run svd2rust?

baldwin1974, 2018-06-10:
Thank you!  Yes, I saw that there is already a crate after I built on for myself. I think this can be closed, thanks for the quick help!

baldwin1974, 2018-06-10:
I guess I should add, for the stm32h7 crate, I can't find any documentation of specific fields/modules/structs. In contrast, the stm32h743 repository can be downloaded and I can run "cargo doc" to get the full API.  I don't suppose you could tell me how to find that?

baldwin1974, 2018-06-10:
Ah, nevermind: https://stm32.agg.io/rs/STM32H7x3.html

adamgreig, 2018-06-10:
You can also clone the [stm32-rs](https://github.com/adamgreig/stm32-rs) repo, build it (run `make`), cd to the stm32h7 crate, and `cargo doc --features stm32h7x3` to do the same. It's a bit cumbersome, it'd be nicer if they were hosted somewhere as per https://github.com/adamgreig/stm32-rs/issues/3 but there's an awful lot of them.

By the way, you can close the issue yourself if you feel it's resolved.

baldwin1974, 2018-06-10:
Ok, I will try that approach to the docs.  Perhaps I should also unpublish stm32h743 crate?

closed 2018-06-10 by baldwin1974