rust-embedded/svd2rust #232: Escape `[` and `]` in svd description before it is used in the doc attribute

Nicoretti, 2018-09-03:
Hi @all,

I generated some peripherals using **svd2rust** and stumbled upon an issue with **missing escape sequences** for characters which should be escaped **in the docstrings**.

**Issue:**
Missing escape sequence for `[` and `]` in generated docstrings.

**Impact:**
Documentation generation will fail.

**Steps to reproduce:**
1. Download [Keil.STM32F4xx_DFP.2.13.0.pack](https://keilpack.azureedge.net/pack/Keil.STM32F4xx_DFP.2.13.0.pack)
   The .pack file actually is a zip file
2. Unzip the .pack file and find the get STM32F41x.svd file
    (CMSIS/SVD/STM32F41x.svd)
3. Run the following svd2rust command
    `svd2rust -i STM32F41x.svd --target "cortex-m"`
4. Add the generated .rs file to some crate
5. Try to generate the documentation for that crate

**Summary**
If an .svd file contains a description with the characters `[` and `]`
e.g.:
```     
    <interrupt>
         <name>EXTI9_5</name>
         <description>EXTI Line[9:5] interrupts</description>
         <value>23</value>
     </interrupt>
```
**svd2rust** will generate a docstring missing necessary escapes for `[` and `]`:

```    
    #[doc = "23 - EXTI Line[9:5] interrupts"]
    EXTI9_5,
```
this will prevent `cargo doc` from generating documentation for the generated file:
```
error: `[9:5]` cannot be resolved, ignoring it...
   --> peripherals/src/lib.rs:259:23
    |
259 |     #[doc = "23 - EXTI Line[9:5] interrupts"]
    |                       ^^^ cannot be resolved, ignoring
    |
note: lint level defined here
   --> peripherals/src/lib.rs:3:12
    |
3   | #![ deny ( warnings ) ]
    |            ^^^^^^^^
    = note: #[deny(intra_doc_link_resolution_failure)] implied by #[deny(warnings)]
    = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`
```

**Attention:**
The suggested cargo/rustc fix will work, but be aware that in order to achiev `\[` you need to write `\\[` I guess this is due to the fact that we need create an escape sequence for the "docstring"  and not for the string itself.

ryankurte, 2018-09-04:
Thanks for the issue and PR! #233 looks like a reasonable solution to me.
 
We seem to be a bit inconsistent with where `util::respace` is called. I wonder whether we should be calling both at all those sites (cc. @jamesmunns)?

Nicoretti, 2018-09-11:
@jamesmunns would it make sense to add the document generation to the regression testing?

jamesmunns, 2018-09-11:
I'll take a look at this (and #233) when I get a chance, but tbh I haven't looked too deeply in how the doc comment generation process works.

@Nicoretti yes, I think that would make sense to add to both svd2rust_regress and the main travis CI.

wez, 2018-09-14:
FWIW, as a crappy but effective workaround, I've been adding `#![allow(intra_doc_link_resolution_failure)]` to the generated crates

commit 35fc5adb (2018-10-16): Added util::escape_brackets function to escape unescpad brackets

... in docstrings. This function takes care to not add another level of
escaping in case a docstring contains already escaped brackets.

Closes #232.

Apologies to @Nicoretti for not noticing his PR earlier. However I do
think it is important to not escape properly escaped brackets again,
this is only meant in cases the author failed to do so.
commit a7fec5bd (2018-10-16): Merge #240

240: Added util::escape_brackets function to escape unescpad brackets r=ryankurte a=therealprof

... in docstrings. This function takes care to not add another level of
escaping in case a docstring contains already escaped brackets.

Closes #232.

Apologies to @Nicoretti for not noticing his PR earlier. However I do
think it is important to not escape properly escaped brackets again,
this is only meant in cases the author failed to do so.
commit 4e999bf1 (2018-10-30): Merge #240 #241

240: Added util::escape_brackets function to escape unescpad brackets r=ryankurte a=therealprof

... in docstrings. This function takes care to not add another level of
escaping in case a docstring contains already escaped brackets.

Closes #232.

Apologies to @Nicoretti for not noticing his PR earlier. However I do
think it is important to not escape properly escaped brackets again,
this is only meant in cases the author failed to do so.

241: Fixed some trivial clippy lints r=ryankurte a=therealprof
commit b7cb4cc3 (2018-10-31): Merge #240

240: Added util::escape_brackets function to escape unescpad brackets r=ryankurte a=therealprof

... in docstrings. This function takes care to not add another level of
escaping in case a docstring contains already escaped brackets.

Closes #232.

Apologies to @Nicoretti for not noticing his PR earlier. However I do
think it is important to not escape properly escaped brackets again,
this is only meant in cases the author failed to do so.
closed 2018-10-31 by therealprof