rust-embedded/svd2rust #249: error: cannot find type `MODE2_ALARM` in this scope

evq, 2018-11-04:
The library generated from a SAM L11 svd fails with the following error:
```
error[E0412]: cannot find type `MODE2_ALARM` in this scope
   --> src/rtc/mod.rs:125:23
    |
125 |     pub mode2_alarm0: MODE2_ALARM,
    |                       ^^^^^^^^^^^ not found in this scope
help: possible candidate is found in another module, you can import it into scope
    |
1   | use rtc::mode2::MODE2_ALARM;
    |
```

Somehow this field in the MODE2 register block fails to reference the `mode2` module like all the others do: 
```
    #[doc = "0x18 - MODE2 Clock Value"]
    pub clock: self::mode2::CLOCK,
    _reserved2: [u8; 4usize],
    #[doc = "0x20 - MODE2 Alarm"]
    pub mode2_alarm0: MODE2_ALARM,
    _reserved3: [u8; 27usize],
    #[doc = "0x40 - General Purpose"]
    pub gp: [self::mode2::GP; 2],
    _reserved4: [u8; 24usize],
```

I noticed that in the svd this seems to be defined as a cluster, rather than a register. Does this help explain the issue?

I am able to workaround through patches.

You can find the svd I used here - https://github.com/evq/atsaml11xxx/blob/master/ATSAML11E16A.orig.svd

jonas-schievink, 2019-11-04:
I ran into the same issue trying to run `svd2rust` on this PSoC 6 SVD: [psoc63.svd.txt](https://github.com/rust-embedded/svd2rust/files/3806111/psoc63.svd.txt)

(upstream repo: https://github.com/cypresssemiconductorco/Cypress-Peripheral-Driver-Library-v3.0.1/blob/master/3.0.1/devices/psoc6/psoc63/psoc63.svd)

jonas-schievink, 2019-11-06:
Ah, both SVDs contain *nested* `<cluster>`s, that must be the reason this fails to compile! (I had no idea that was even possible)

burrbull, 2019-11-08:
I think the bug is somewhere in https://github.com/rust-embedded/svd2rust/blob/11e11f0498c0aca1042eb927e2aa74c246a8b4a0/src/generate/register.rs#L755

jonas-schievink, 2019-11-26:
Also seems to happen with the nRF5340 SVDs, unfortunately

commit 6594018b (2021-02-04): Fix type path for cluster fields. Allows nested clusters

Fixes #249
commit cc453135 (2021-02-04): Merge #489

489: Fix type path for cluster fields. Allows nested clusters r=Emilgardis a=theli-ua

Fixes #249

Tested on following test svd: https://gist.github.com/theli-ua/a1b9c17e94442fbd56c774bb504795a0
PR #489 (2021-02-04): Fix type path for cluster fields. Allows nested clusters
Fixes #249

Tested on following test svd: https://gist.github.com/theli-ua/a1b9c17e94442fbd56c774bb504795a0
closed 2021-02-04 by bors[bot]