rust-embedded/svd2rust #250: libssl1.1.1

nickray, 2018-11-04:
Running on debian/testing is turning a lot of crates into a bit of a saga currently...

To build `svd2rust-regress`, what worked for me (consulting https://github.com/sfackler/rust-openssl/issues/994#issuecomment-429528860) is adding
```toml
[replace]
"openssl:0.9.24" = { git = "https://github.com/ishitatsuyuki/rust-openssl", branch = "0.9.x" }
```
to its `Cargo.toml`.

What is the proper way to get this to run?

jamesmunns, 2018-11-05:
@nickray I noticed this, but haven't pushed an update for this.

I expect to bump our dependency on `reqwest` from 0.8 to 0.9, which will pull in the newest version of `openssl`. If this change works for you, and if you'd like to submit this change as a PR, I'll merge it quickly!

adamgreig, 2019-07-28:
This should be fixed by #346.

closed 2019-07-28 by adamgreig