rust-embedded/svd2rust #259: Panic processing SVD file calling `c.text.clone()`

thejpster, 2018-11-20:
I have received an SVD file from Nuvoton for the M2351 (I'm trying to work out if I can publish it). If I put it through svd2rust, this happens:

```
$ RUST_BACKTRACE=1 ../svd2rust/target/debug/svd2rust.exe -i ./M2351_v1.svd
thread 'main' panicked at 'C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:60 c.text.clone()', libcore\option.rs:1000:5
stack backtrace:
   0: std::sys::windows::backtrace::set_frames
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\sys\windows\backtrace\mod.rs:104
   1: std::sys::windows::backtrace::set_frames
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\sys\windows\backtrace\mod.rs:104
   2: std::sys_common::backtrace::_print
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\sys_common\backtrace.rs:71
   3: std::sys_common::backtrace::_print
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\sys_common\backtrace.rs:71
   4: std::panicking::default_hook::{{closure}}
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:211
   5: std::panicking::default_hook
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:227
   6: std::panicking::rust_panic_with_hook
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:477
   7: std::panicking::continue_panic_fmt
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:391
   8: std::panicking::rust_begin_panic
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:326
   9: core::panicking::panic_fmt
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libcore\panicking.rs:77
  10: core::option::expect_failed
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libcore\option.rs:1000
  11: core::option::Option<alloc::string::String>::expect<alloc::string::String>
             at \libcore\option.rs:312
  12: svd_parser::{{impl}}::get_child_text::{{closure}}<str*>
             at C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:60
  13: core::option::Option<xmltree::Element*>::map<xmltree::Element*,alloc::string::String,closure>
             at \libcore\option.rs:414
  14: svd_parser::{{impl}}::get_child_text<str*>
             at C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:60
  15: svd_parser::Field::parse
             at C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:449
  16: core::ops::function::FnMut::call_mut<fn(xmltree::Element*) -> svd_parser::Field,(xmltree::Element*)>
             at \libcore\ops\function.rs:156
  17: core::ops::function::impls::{{impl}}::call_once<(xmltree::Element*),fn(xmltree::Element*) -> svd_parser::Field>
             at \libcore\ops\function.rs:286
  18: core::option::Option<xmltree::Element*>::map<xmltree::Element*,svd_parser::Field,mut fn(xmltree::Element*) -> svd_parser::Field*>
             at \libcore\option.rs:414
  19: core::iter::{{impl}}::next<svd_parser::Field,core::slice::Iter<xmltree::Element>,fn(xmltree::Element*) -> svd_parser::Field>
             at \libcore\iter\mod.rs:1394
  20: alloc::vec::{{impl}}::spec_extend<svd_parser::Field,core::iter::Map<core::slice::Iter<xmltree::Element>, fn(xmltree::Element*) -> svd_parser::Field>>
             at \liballoc\vec.rs:1903
  21: alloc::vec::{{impl}}::from_iter<svd_parser::Field,core::iter::Map<core::slice::Iter<xmltree::Element>, fn(xmltree::Element*) -> svd_parser::Field>>
             at \liballoc\vec.rs:1886
  22: alloc::vec::{{impl}}::from_iter<svd_parser::Field,core::iter::Map<core::slice::Iter<xmltree::Element>, fn(xmltree::Element*) -> svd_parser::Field>>
             at \liballoc\vec.rs:1772
  23: core::iter::iterator::Iterator::collect<core::iter::Map<core::slice::Iter<xmltree::Element>, fn(xmltree::Element*) -> svd_parser::Field>,alloc::vec::Vec<svd_parser::Field>>
             at \libcore\iter\iterator.rs:1415
  24: svd_parser::{{impl}}::parse::{{closure}}
             at C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:366
  25: core::option::Option<xmltree::Element*>::map<xmltree::Element*,alloc::vec::Vec<svd_parser::Field>,closure>
             at \libcore\option.rs:414
  26: svd_parser::RegisterInfo::parse
             at C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:365
  27: svd_parser::Register::parse
             at C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:392
  28: svd_parser::cluster_register_parse
             at C:\Users\jgp\.cargo\registry\src\github.com-1ecc6299db9ec823\svd-parser-0.6.0\src\lib.rs:264
  29: core::ops::function::FnMut::call_mut<fn(xmltree::Element*) -> either::Either<svd_parser::Register, svd_parser::Cluster>,(xmltree::Element*)>
             at \libcore\ops\function.rs:156
  30: core::ops::function::impls::{{impl}}::call_once<(xmltree::Element*),fn(xmltree::Element*) -> either::Either<svd_parser::Register, svd_parser::Cluster>>
             at \libcore\ops\function.rs:286
  31: core::option::Option<xmltree::Element*>::map<xmltree::Element*,either::Either<svd_parser::Register, svd_parser::Cluster>,mut fn(xmltree::Element*) -> either::Either<svd_parser::Register, svd_parser::Cluster>*>
             at \libcore\option.rs:414
[... 39 lines cut ...]
             at C:\Users\jgp\Documents\svd2rust\src\main.rs:125
  52: std::rt::lang_start::{{closure}}<()>
             at \libstd\rt.rs:74
  53: std::rt::lang_start_internal::{{closure}}
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\rt.rs:59
  54: std::rt::lang_start_internal::{{closure}}
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\rt.rs:59
  55: panic_unwind::__rust_maybe_catch_panic
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libpanic_unwind\lib.rs:103
  56: std::panicking::try
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:289
  57: std::panicking::try
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:289
  58: std::panicking::try
             at /rustc/1433507eba7d1a114e4c6f27ae0e1a74f60f20de\src/libstd\panicking.rs:289
  59: std::rt::lang_start<()>
             at \libstd\rt.rs:74
  60: main
  61: invoke_main
             at f:\dd\vctools\crt\vcstartup\src\startup\exe_common.inl:78
  62: invoke_main
             at f:\dd\vctools\crt\vcstartup\src\startup\exe_common.inl:78
  63: BaseThreadInitThunk
  64: RtlUserThreadStart
```

thejpster, 2018-11-20:
I now have permission from Nuvoton to publish the SVD file but it remains unclear what the license terms are.

[M2351_v1.zip](https://github.com/rust-embedded/svd2rust/files/2599539/M2351_v1.zip)

kamathba, 2018-11-24:
I get a similar error with svd files for ATSAML10xx.svd and ATSAML11xx.svd files. It's strange though, since one of the svd files I got from here: https://github.com/evq/atsaml11xxx and it looks like @evq was able to run it at some point. I'm pretty inexperienced with svd2rust though so I could be doing something wrong.

```
thread 'main' panicked at '/home/bkamath/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.6.0/src/lib.rs:60 c.text.clone()', src/libcore/option.rs:1008:5
stack backtrace:
   0: std::sys::unix::backtrace::tracing::imp::unwind_backtrace
             at src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1: std::sys_common::backtrace::_print
             at src/libstd/sys_common/backtrace.rs:71
   2: std::panicking::default_hook::{{closure}}
             at src/libstd/sys_common/backtrace.rs:59
             at src/libstd/panicking.rs:211
   3: std::panicking::default_hook
             at src/libstd/panicking.rs:227
   4: std::panicking::rust_panic_with_hook
             at src/libstd/panicking.rs:476
   5: std::panicking::continue_panic_fmt
             at src/libstd/panicking.rs:390
   6: rust_begin_unwind
             at src/libstd/panicking.rs:325
   7: core::panicking::panic_fmt
             at src/libcore/panicking.rs:77
   8: core::option::expect_failed
             at src/libcore/option.rs:1008
   9: svd_parser::cluster_register_parse
  10: <alloc::vec::Vec<T> as alloc::vec::SpecExtend<T, I>>::from_iter
  11: svd_parser::Peripheral::parse
  12: <alloc::vec::Vec<T> as alloc::vec::SpecExtend<T, I>>::from_iter
  13: svd_parser::Device::parse
  14: svd_parser::parse
  15: svd2rust::main
  16: std::rt::lang_start::{{closure}}
  17: std::panicking::try::do_call
             at src/libstd/rt.rs:59
             at src/libstd/panicking.rs:310
  18: __rust_maybe_catch_panic
             at src/libpanic_unwind/lib.rs:102
  19: std::rt::lang_start_internal
             at src/libstd/panicking.rs:289
             at src/libstd/panic.rs:398
             at src/libstd/rt.rs:58
  20: main
  21: __libc_start_main
  22: _start
```
Command ran `RUST_BACKTRACE=1 svd2rust -i ATSAML11E16A.orig.svd`

evq, 2018-11-24:
Ah @kamathba that's because the original SVD is malformed. I distributed the original SVD as is and a patch series that I used to fix the issues so svd2rust can run - https://github.com/evq/atsaml11xxx/blob/master/patches/svd-compat.diff

In my particular case the error was due to the single empty description tags: `<description/>` which is against the SVD spec as I understand it. See https://github.com/japaric/svd/issues/59

If you run `make ATSAML11E16A.svd` it will apply that patchset using `quilt`.

kamathba, 2018-11-24:
Ah, gotcha! The SVDConv tool distributed with the CMSIS pack doesn't seem to catch this as a problem. It would be nice if svd2rust handled schema/format issues in a more straightforward way (an option or another distributed another binary like SVDConv or something that could point out the issues svd2rust will care about)

burrbull, 2019-07-25:
> I now have permission from Nuvoton to publish the SVD file but it remains unclear what the license terms are.
> 
> [M2351_v1.zip](https://github.com/rust-embedded/svd2rust/files/2599539/M2351_v1.zip)

```
Expected content in <description> tag, found none }
In field `INTFLAG` }
Parsing field #8 }
In register `FMC_ISPSTS` }
In peripheral `FMC` }', src/libcore/result.rs:1051:5
```

burrbull, 2022-11-07:
Closing as should be fixed. Reopen if it is not.

closed 2022-11-07 by burrbull