rust-embedded/svd2rust #275: const fn?

cbiffle, 2019-01-27:
I notice that `const fn` doesn't see a lot of use in the generated code. I suspect this is to retain compatibility with older versions of Rust. Now that the feature is stable, there are a bunch of areas of the API that would benefit from it.

I maintain a [very similar library for C++11](https://github.com/cbiffle/etl/blob/master/biffield/README.mkdn) that heavily relies on `constexpr`, and there are things I can do there that I just can't do with the svd2rust APIs.

For starters: all the `W::reset_value()` functions would make excellent `const fn`s.

In my ideal world, the builder-style operations on `W` would also be `const fn`. This would allow me to write code like this C++ example (taken from [my graphics library](https://github.com/cbiffle/m4vgalib/blob/master/vga.cc#L86)):

```c++
// Common fields used in scanout DMA transfer settings.
static constexpr auto dma_xfer_common = Dma::Stream::cr_value_t()
  .with_chsel(6)  // for TIM1_UP
  .with_pl(Dma::Stream::cr_value_t::pl_t::very_high)
  .with_pburst(Dma::Stream::BurstSize::single)
  .with_mburst(Dma::Stream::BurstSize::single)
  .with_en(true);
```

However, the current builder API can't be easily const-fn'd because of the (strange, imho) decision to pass around `&mut W` instead of just `W` by value. The current stable `const fn` support won't allow the use of mutable references like that. I actually think working on `W` by-value would have some ergonomics improvements, but it'd likely be a significant breaking change.

japaric, 2019-02-08:
Note that `const fn` does *not* mean this function call will *always* be
evaluated at compile time. It only means that the function can be called in
"const context": that is to initialize `const` and `static` items. At least
that's the case today. So unless you are willing to write code like this:

``` rust
const VALUE: BigStruct = some_const_fn();

let x = some_non_const_fn(VALUE);
```

instead of this

``` rust
let x = some_non_const_fn(some_const_fn());
```

you will not see any execution time improvements in *unoptimized* code -- in
optimized code llvm const propagation pass will take care of evaluating most
things at compile time so using `const fn` outside "const context" has no
advantages in optimized code.

For this reason, most libraries only mark their constructors as `const fn`.

Could you say more about how you plan to use a const-ified svd2rust API?

>  The current stable `const fn` support won't allow the use of mutable
>  references like that.

Current stable `const fn` also doesn't have trait bounds like `F: FnOnce()` so
anything involving closures can't be const-ified today.

cbiffle, 2019-03-12:
I'm not trying to use `const fn` to speed up execution time, but to create compile-time constants. The example I posted in C++ is a simple case, but more broadly: I have applications that store peripheral configurations as data structures. I frequently want these data structures to go into ROM, and so I need to write them as a `static` currently.

[Here is a concrete example in Rust.](https://github.com/cbiffle/m4vga-rs/blob/c5f749e3cad31c05f4ed007684a499eeec8e417c/src/util/stm32.rs#L9). I need to refer to the divisor settings for each output tap of the PLL in this struct. I could create my own enum and translate it to and from the svd2rust one, but it's more convenient to just use the svd2rust one.

I [create an instance of that struct in ROM](https://github.com/cbiffle/m4vga-rs/blob/c5f749e3cad31c05f4ed007684a499eeec8e417c/src/timing.rs#L86) using the enum variants, which I can do for enums. But if I were instead trying to store the value of an entire register, because of the lack of `const fn reset_value()` and modifiers on the current register types, I would basically need to store a `u32` that I have manually ensured matches the desired register contents.

Contrast with my C++ API that I quoted above, where the register defaults and modifiers are `constexpr` and can be used to create const static structures.

Does that make sense?

(Re: the issue with trait bounds and closures -- yes, I think the closure-based modification APIs on the current svd2rust generated register types are probably a poor fit for this case. Unfortunately there is no alternative at the moment.)

therealprof, 2019-07-25:
@cbiffle In v0.15.0 `reset_value()` and `ptr()` are now const fn.

closed 2019-07-25 by therealprof