rust-embedded/svd2rust #276: Prepare for new release

therealprof, 2019-01-29:
Quite a few PRs and issues and interesting ideas have accumulated. So the big questions are: What do we want to land, when, and is it going to be in a minor release or do we want a patch release first?

CC @rust-embedded/tools

japaric, 2019-02-05:
Changes that have landed since v0.14.0

- #269 ci: remove rustfmt for speed
- #214 Provide library access to code generation
- #274 Implement Copy, Clone, Debug for Interrupt

None of them warrant a minor version bump.

burrbull, 2019-07-21:
I think it's time for new release.
As soon as #324 be merged.

CC @rust-embedded/tools

burrbull, 2019-07-21:
Draft of CHANGELOG:

- Logging system was introduced by `log` crate.

- `svd2rust` can be used as library.

- `derive_from` now can be used for registers.

- [breaking-change] for access to alternate registers functions now used
  instead of untagged_unions (no more nightly `features`)

- generated code now more compact and compilation faster

- `reset_value` now public static method of register structure

- `Clone`, `Copy`, `Debug`, `PartialEq` implemented for read/write enums

cc @therealprof

therealprof, 2019-07-21:
@burrbull I agree. Thanks a lot for your contributions, much appreciated! Maybe you may want to consider becoming part of the tools team?

burrbull, 2019-07-21:
If you insist. :)

therealprof, 2019-07-21:
The tools team could definitely use some more manpower and people willing to contribute time. You've shown a lot of passion the last weeks and moved a lot forward including by taking on stale PRs and freshen them up. You being a team member definitely has my endorsement. You might want to check https://github.com/rust-embedded/wg/blob/master/rfcs/0136-teams.md#adding-new-members for details about the process.

burrbull, 2019-07-21:
Is it needed to open inviting PR by myself? It looks strange for me.

therealprof, 2019-07-21:
That's how it works. You create a PR proposing to add yourself and then the team members have to approve in order to land the PR.

therealprof, 2019-07-25:
@burrbull I think the release is due now. Want to turn your proposed changes into a PR so we can get this on the road?

burrbull, 2019-07-25:
Let's make release as is.

therealprof, 2019-07-25:
Someone needs to crate a PR with the changelog and version bump. 😅 Since you already drafted the CHANGELOG here, would you like to do a PR with the changes? If not I'll be happy to do it.

burrbull, 2019-07-25:
Ok. Do I need to change something in changelog?

therealprof, 2019-07-25:
Your proposal sounds good to me. I haven't checked for completeness though.

burrbull, 2019-07-25:
#336

therealprof, 2019-07-25:
This has happened. 🎉

closed 2019-07-25 by therealprof