rust-embedded/svd2rust #296: Warn or fail on unknown tags?

laanwj, 2019-05-20:
I had a typo here, specifying `enumerateValue` instead of `enumeratedValue`, and the parser silently ignored it: https://github.com/riscv-rust/k210-pac/pull/23
Would be convenient if these unknown tags were flagged with a warning at least.

Disasm, 2019-05-20:
Maybe this issue should be addressed to the parser: https://github.com/japaric/svd

laanwj, 2019-05-20:
Yes, I'm not really familiar with the inner working of this tool, but that makes sense, the parser would at least need to report this to be able to report this to the user

Disasm, 2019-05-22:
Filed a PR for this: https://github.com/japaric/svd/pull/69
However, it's just for `enumeratedValues`, maybe the same problem can happen again somewhere.

closed 2022-04-14 by laanwj