rust-embedded/svd2rust #299: Incorrect handling of cluster `size` field

burrbull, 2019-05-23:
size | Define the default bit-width of any device register (implicit inheritance).

It's size of each register in cluster, not a size of whole cluster.

https://github.com/riscv-rust/k210-pac/pull/24

Emilgardis, 2019-05-29:
I think this is where we wrongly use `cluster_size` for determining sequentiality. https://github.com/rust-embedded/svd2rust/blob/20d8bd4c7554c5b9ced6c1c6941a81747274e9e4/src/generate/peripheral.rs#L614-L644

I'm not sure how this has not tripped anything earlier, thanks for reporting it 👍 
The fix should be easy, I'll see what I can do. Interested in how many svds this is effected by.

burrbull, 2019-05-30:
I think first of all you should ignore `size` field here https://github.com/rust-embedded/svd2rust/blob/20d8bd4c7554c5b9ced6c1c6941a81747274e9e4/src/generate/peripheral.rs#L601 and always use `cluster_size_in_bits`:
`let cluster_size = cluster_size_in_bits(cluster, defs).ok_or_else(...);`
This should fix most of broken SVDs.

In my view `size` should be argument of `parse` and be passed down the chain from device to register.
But it will require changes in `svd-parser`.

commit c0208332 (2019-07-21): Merge #325

325: fix cluster size r=therealprof a=burrbull

Fix misusing cluster size tag.

Close #299 .

r? @therealprof
PR #325 (2019-07-21): fix cluster size
Fix misusing cluster size tag.

Close #299 .

r? @therealprof
closed 2019-07-21 by bors[bot]