rust-embedded/svd2rust #304: why is this "tool" not working

josephpenafiel, 2019-07-06:
I'm getting this error:

thread 'main' panicked at '/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.6.0/src/lib.rs:361 parse::u32(t)', src/libcore/option.rs:1034:5
note: Run with `RUST_BACKTRACE=1` environment variable to display a backtrace.

tried installing svd-parser, and get this error: 
error: specified package has no binaries

How is this tool supposed to be used? there's little documentation on this. I need help

burrbull, 2019-07-06:
When does this error occur? When you install svd2rust?
```
cargo install svd2rust
```

josephpenafiel, 2019-07-06:
when I run:
$ svd2rust -i STM32G07x.svd --target cortex-m

burrbull, 2019-07-06:
Can you store somewhere you svd file?

And did you try [stm32g4xx-hal](https://crates.io/crates/stm32g4xx-hal) crate?

josephpenafiel, 2019-07-06:
https://1drv.ms/u/s!AsXdvpVk4QASkI8moO6F1uGJPT0SAg

burrbull, 2019-07-06:
You can try svd2rust from master branch:
```
cargo install svd2rust --get https://github.com/rust-embedded/svd2rust --force
```

josephpenafiel, 2019-07-06:
Tried your suggestion but I'm still getting the error

burrbull, 2019-07-06:
```
thread 'main' panicked at '//TODO(AJM): SVDError { inner: SVDError { inner: SVDError { inner: ParseIntError { kind: Overflow }

0x1F1F1F11F invalid }

In register `HWCFGR6` }

In peripheral `ADC` }',
```

adamgreig, 2019-07-26:
Closing, this is a bug in the svd file rather than svd2rust. @josephpenafiel please feel free to comment again if you still can't get it working, but also consider using the stm32g0 crate.

closed 2019-07-26 by adamgreig