rust-embedded/svd2rust #351: Invalid Enum values are permitted

axos88, 2019-07-29:
The following will generate code, although the value is invalid, since it doesn't fit into the number of bits described:

```
<field>
	<name>AADEN</name>
	<description>AAD enable.</description>
	<bitRange>[2:2]</bitRange>
	<enumeratedValues>
		<name>ENUM</name>
		<enumeratedValue>
			<name>AUTO_ADDRESS_DETECT_ON</name>
			<description>Auto Address Detect (AAD) is disabled.</description>
			<value>122213</value>
		</enumeratedValue>
		<enumeratedValue>
			<name>AUTO_ADDRESS_DETECT_OFF</name>
			<description>Auto Address Detect (AAD) is enabled.</description>
			<value>49584798</value>
		</enumeratedValue>
	</enumeratedValues>
</field>
```

burrbull, 2019-07-30:
Make issue in https://github.com/japaric/svd also, please.

burrbull, 2022-11-07:
should be fixed

closed 2022-11-07 by burrbull