rust-embedded/svd2rust #352: Enum with values named identically generates invalid code

axos88, 2019-07-29:
The following generates invalid code:

```
<field>
	<name>AADEN</name>
	<description>AAD enable.</description>
	<bitRange>[2:2]</bitRange>
	<enumeratedValues>
		<name>ENUM</name>
		<enumeratedValue>
			<name>AUTO_ADDRESS_DETECT_</name>
			<description>Auto Address Detect (AAD) is disabled.</description>
			<value>1</value>
		</enumeratedValue>
		<enumeratedValue>
			<name>AUTO_ADDRESS_DETECT_</name>
			<description>Auto Address Detect (AAD) is enabled.</description>
			<value>0</value>
		</enumeratedValue>
	</enumeratedValues>
</field>
```

```
#[doc = "Possible values of the field `AADEN`"]
#[derive(Clone, Copy, Debug, PartialEq)]
pub enum AADENR {
    #[doc = "Auto Address Detect (AAD) is disabled."]
    AUTO_ADDRESS_DETECT_,
    #[doc = "Auto Address Detect (AAD) is enabled."]
    AUTO_ADDRESS_DETECT_,
}
```

axos88, 2019-07-30:
In such cases shouldn't we automatically rename the fields to AUTO_ADDRESS_DETECT_ON / OFF? And issue a warning at least?

therealprof, 2019-07-30:
How exactly would you know that the missing pieces in the identifier are `ON` and `OFF`? Could also be `HOT` and `COLD`, `YES` and `NO`, `FOO` and `BAR`.

> And issue a warning at least?

This will not compile but of course PRs to implement better diagnostics are always welcome.

axos88, 2019-07-30:
Well "it's something"... Still better than generating invalid code, isn't it? We could use the values instead of on / off to be general

therealprof, 2019-07-30:
I'm very much opposed to introducing automagic in attempt to fix broken SVD files. Once you start going down that rabbit hole you'll find lots of broken things you might want want to autocorrect.

closed 2019-07-30 by axos88