rust-embedded/svd2rust #357: Duplicated identifiers generated

axos88, 2019-07-30:
The SVD file below generates two definitions of the struct `PIO0_`. This seems like a tricky edge case where the repetition is 0-3 + 6-32 with the same type, and exceptions for repetitions 4 and 5. The types for 0-3 and 6-32 are the same, so we *should* be able to reuse that?

Also I'm baffled by the generated code, why some element generate an array, and others an entry for each repetition? Specifically why the code generated by the first and fourth register definition differs, since the definition seems to be exactly the same except offsets and repetition numbers.
```
    pub pio0__: [PIO0__; 4],   // Generated by the first register definition with repetition
    #[doc = "0x10 - I/O configuration for open-drain pin PIO0_4"]
    pub pio0_4: PIO0_4,    // Generated by the second register definition without repetition
    #[doc = "0x14 - I/O configuration for open-drain pin PIO0_5"]
    pub pio0_5: PIO0_5,    // Generated by the second register definition without repetition
    #[doc = "0x18 - I/O configuration for port PIO0"]
    pub pio0_6: PIO0_, // Generated by the fourth register definition with repetition
    #[doc = "0x1c - I/O configuration for port PIO0"]
    pub pio0_7: PIO0_,  // Generated by the fourth register definition with repetition
    #[doc = "0x20 - I/O configuration for port PIO0"]
    pub pio0_8: PIO0_,  // Generated by the fourth register definition with repetition
    #[doc = "0x24 - I/O configuration for port PIO0"]
    pub pio0_9: PIO0_,  // Generated by the fourth register definition with repetition

```

```
<peripheral>
	<name>IOCON</name>
	<description>I/O control (IOCON) </description>
	<groupName>IOCON</groupName>
	<baseAddress>0x40044000</baseAddress>
	<addressBlock>
	<offset>0x0</offset>
	<size>0xFFF</size>
	<usage>registers</usage>
	</addressBlock>
	<registers>
		<register>
			<dim>4</dim>
			<dimIncrement>0x4</dimIncrement>
			<dimIndex>0-3</dimIndex>
			<name>PIO0_%s</name>
			<description>I/O configuration for  port PIO0</description>
			<addressOffset>0x000</addressOffset>
			<access>read-write</access>
			<resetValue>0x00000090</resetValue>
			<resetMask>0xFFFFFFFF</resetMask>
			<fields>
				<field>
					<name>FUNC</name>
					<description>Selects pin function.</description>
					<bitRange>[2:0]</bitRange>

				</field>
				<field>
					<name>MODE</name>
					<description>Selects function mode (on-chip pull-up/pull-down resistor control).</description>
					<bitRange>[4:3]</bitRange>
					<enumeratedValues>
					<name>ENUM</name>
						<enumeratedValue>
							<name>INACTIVE_NO_PULL_DO</name>
							<description>Inactive (no pull-down/pull-up resistor enabled).</description>
							<value>0x0</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>PULL_DOWN_RESISTOR_E</name>
							<description>Pull-down resistor enabled.</description>
							<value>0x1</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>PULL_UP_RESISTOR_ENA</name>
							<description>Pull-up resistor enabled.</description>
							<value>0x2</value>
						</enumeratedValue>
[... 972 lines cut ...]
							<value>0x4</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>IOCONCLKDIV5</name>
							<description>IOCONCLKDIV5. Use IOCON clock divider 5.</description>
							<value>0x5</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>IOCONCLKDIV6</name>
							<description>IOCONCLKDIV6. Use IOCON clock divider 6.</description>
							<value>0x6</value>
						</enumeratedValue>
					</enumeratedValues>
				</field>
				<field>
					<name>RESERVED</name>
					<description>Reserved.</description>
					<bitRange>[31:16]</bitRange>

				</field>
			</fields>
		</register>
	</registers>

```

burrbull, 2019-07-30:
Github have gist: https://gist.github.com/

svd2rust creates array only if first item has index `0`.

axos88, 2019-07-31:
Ah gotcha, that makes sense, although it is weird in this case. Is there a way to make it *not* generate an array?

Also this is an inconvenience, and odd, but the real problem is the duplicated identifier being generated.

burrbull, 2019-07-31:
> Is there a way to make it _not_ generate an array

We can add this as option.

But in your case PIO0_4 and PIO0_5 differ from others so all works as expected.

axos88, 2019-07-31:
The problem is that PIO0-3 and PIO6-31 have the same type, and because of this the PIO_ struct is generated twice (although identically), and the crate fails to build.
Also PIO0-3 is an array, while PIO6-31 is generated separately, and PIO4,5 have different types.

burrbull, 2019-07-31:
I see only one way: rename `PIO_%s` 6-31. Then they will have other type.

As option you can `<deriveFrom>` this from first part to use same type, but I think this don't work with arrays yet (need to check).

axos88, 2019-07-31:
Yeah that's the workaround for now, but that makes accessing those registers even more inconsistent. rust2svd should either understand that the two definitions for PIO%s are identical, and not generate the structs twice, or the dimIndex should accept non-continous blocks, but I guess that's not part of the SVD specification, so that's a no go.

n8tlarsen, 2022-11-07:
Actually your issue already points out some room for improvement in that not only must the register name regex match but the fields as well before determining that the types should be implicitly derived.

burrbull, 2022-11-07:
> Actually your issue already points out some room for improvement in that not only must the register name regex match but the fields as well before determining that the types should be implicitly derived.

Exactly.

n8tlarsen, 2022-11-07:
Should I try to fix this before merging #662 or open a new PR?

burrbull, 2022-11-07:
Closed by #662

closed 2022-11-07 by burrbull