rust-embedded/svd2rust #366: New release

burrbull, 2019-08-04:
cc @therealprof 

I think it is time for new release.

therealprof, 2019-08-04:
I haven't checked the impact of the changes yet. Except for the unsafe fix, what do you figure is important enough to warrant a new release? And do you mean before or after #364 (which lacks the CHANGELOG entry, BTW).

burrbull, 2019-08-04:
I think it's better to have release before 364.

That PR is serious change and we can miss a lot of bugs.

therealprof, 2019-08-04:
Works for me. Would you mind re-checking all details (like CHANGELOG entries) and create a PR?

burrbull, 2019-08-04:
15.3 or 16.0?

therealprof, 2019-08-04:
The `-g` flag is a breaking change, right?

therealprof, 2019-08-04:
Ah, I see, the default hasn't changed. But curiously the `-g` flag is not mentioned in the CHANGELOG. 😅

therealprof, 2019-08-04:
Still the new methods in the generated code probably warrant a minor bump.

burrbull, 2019-08-04:
> Still the new methods in the generated code probably warrant a minor bump.

Yes, `-g` is needed to simplify the change-over.

>  But curiously the `-g` flag is not mentioned in the CHANGELOG. sweat_smile

I remember.

burrbull, 2019-08-04:
> The `-g` flag is a breaking change, right?

By itself, no. You can to not use it and you'll not see changes.

closed 2019-08-05 by burrbull