rust-embedded/svd2rust #368: Writing part of register

burrbull, 2019-08-08:
cc @therealprof 
cc @Disasm 

Solution option:
```rust
pub struct _W;
impl<U> Writable for Reg<U, _W> {}

impl<REG> Reg<u32, REG>
where
    Self: Writable
{
    pub unsafe fn u16(&self) -> &[Reg<u16, _W>; 2] {
        &*((self as *const Self) as *const [Reg<u16, _W>; 2])
    }
    pub unsafe fn u8(&self) -> &[Reg<u8, _W>; 4] {
        &*((self as *const Self) as *const [Reg<u8, _W>; 4])
    }
}
```
Same for 64-bit registers.

Usage:
```rust
unsafe { reg.u8()[3].write_with_zero(|w| w.bits(15)); }
```

Disasm, 2019-08-08:
I do not think this is a good idea in general. These methods are inside a generic type, so you can accidentally call them for registers that do not support half-word access. But if you can generate these functions only for specific registers, then it would be safer.

burrbull, 2019-08-08:
I can, but where do I get information which registers are "specific". SVD doesn't contain it.

Disasm, 2019-08-08:
Hmm. Are there any examples of such registers, at least in STM32?

closed 2021-04-17 by burrbull