rust-embedded/svd2rust #376: Better error messages - revisited

nickray, 2019-08-14:
In some cases, upgrading from 0.14.0 to 0.16.0 seems to reduce the error message quality. I'm trying to `svd2rust` the ARM Musca-B1 SVD, which has bugs. Not quite sure the license or I would attach the SVD.

Would it be possible to output the SVD line number that causes the error?

## Case 1
0.14.0:
`thread 'main' panicked at 'unknown access variant: read', /home/nicolas/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.6.0/src/lib.rs:426:18`

0.16.0:
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: SVDError { inner: SVDError { inner: SVDError { inner: 
unknown access variant found }
In register `SPIISR` }
In peripheral `SPI0` }', src/libcore/result.rs:999:5
```

## Case 2
0.14.0: `thread 'main' panicked at '/home/nicolas/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.6.0/src/lib.rs:354 tree.get_child_text("description")', src/libcore/option.rs:1008:5`

0.16.0: (with `-l trace`): `thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', src/libcore/option.rs:347:21`

therealprof, 2019-08-14:
Case 2 sounds like it might be addressed by #369.

Indeed the error messages could be much better; hopefully we'll be able to address those things in the future with the upgrade to the latest syn/quote versions and a better preprocessing of the SVD trees.

nickray, 2019-08-14:
Oh, perfect, thanks for the pointer! For case 1 I put together https://github.com/japaric/svd/pull/75.

closed 2019-08-14 by nickray