rust-embedded/svd2rust #378: Derived peripheral not quite working

nickray, 2019-08-14:
I understand https://github.com/rust-embedded/svd2rust/pull/319 implements derived peripherals.
In my ARM Musca-B1 SVD, the derivation isn't quite working. Should it in this case?

Error for timer1 (derived from timer0):
```
error[E0433]: failed to resolve: use of undeclared type or module `timer1`
    --> src/lib.rs:1676:270
     |
1676 | # [ doc = "Timer 1 (Secure)" ] pub struct TIMER1_SECURE { _marker : PhantomData < * const ( ) > } unsafe impl Send for TIMER1_SECURE { } impl TIMER1_SECURE { # [ doc = r"Returns a pointer to the register block" ] # [ inline ( always ) ] pub const fn ptr ( ) -> * const timer1 :: RegisterBlock { 0x5000_1000 as * const _ } } impl Deref for TIMER1_SECURE { type Target = timer1 :: RegisterBlock ; fn deref ( & self ) -> & Self :: Target { unsafe { & * TIMER1_SECURE :: ptr ( ) } } }
     |                                                                                                                                                                                                                                                                              ^^^^^^ use of undeclared type or module `timer1`
```

SVD excerpts causing the issue:
[excerpt-timer.txt](https://github.com/rust-embedded/svd2rust/files/3503091/excerpt-timer.txt)
[excerpt-uart.txt](https://github.com/rust-embedded/svd2rust/files/3503092/excerpt-uart.txt)

burrbull, 2019-08-15:
No #319 is about derived registers.

And error is in `TIMER1_SECURE` register. Not in peripheral.

burrbull, 2021-04-17:
Any changes?
Please, place full SVD to test it.

nickray, 2021-04-22:
I'm not sure whether this ended up working, I did not end up using the board.
Repo is: https://github.com/nickray/muscab1-pac
With current version svd patch complains about `svdtools.patch.UnknownTagError: licenseText`.

Closing as no longer relevant to me.

closed 2021-04-22 by nickray