rust-embedded/svd2rust #385: svd2rust does not generate compilable code for SAMR34J18B

srenauld, 2019-08-30:
The [following SVD](https://gist.github.com/srenauld/e3b0ab2cd435ae0a710266e2925d01ae) for Microchip's serie of LoRa all-in-one chips produces uncompilable code.

The first error is the use of `ASYNC` as an `enum` in `lib.rs`. This is easily fixed by replacing all instances with `r#ASYNC`.

A subsequent error prevents me from getting any further with this, however. The following structs are generated in `src/gclk/syncbusy.rs` (72 errors total):

```
impl From<GENCTRL0_A> for bool {
    #[inline(always)]
    fn from(variant: GENCTRL0_A) -> Self {
        match variant {
            GENCTRL0_A::GCLK0 => true,
            GENCTRL0_A::GCLK1 => true,
            GENCTRL0_A::GCLK2 => true,
            GENCTRL0_A::GCLK3 => true,
            GENCTRL0_A::GCLK4 => true,
            GENCTRL0_A::GCLK5 => true,
            GENCTRL0_A::GCLK6 => true,
            GENCTRL0_A::GCLK7 => true,
            GENCTRL0_A::GCLK8 => true,
        }
    }
}
#[doc = "Reader of field `GENCTRL0`"]
pub type GENCTRL0_R = crate::R<bool, GENCTRL0_A>;
impl GENCTRL0_R {
    #[doc = r"Get enumerated values variant"]
    #[inline(always)]
    pub fn variant(&self) -> crate::Variant<bool, GENCTRL0_A> {
        use crate::Variant::*;
        match self.bits {
            true => Val(GENCTRL0_A::GCLK0),
            true => Val(GENCTRL0_A::GCLK1),
            true => Val(GENCTRL0_A::GCLK2),
            true => Val(GENCTRL0_A::GCLK3),
            true => Val(GENCTRL0_A::GCLK4),
            true => Val(GENCTRL0_A::GCLK5),
            true => Val(GENCTRL0_A::GCLK6),
            true => Val(GENCTRL0_A::GCLK7),
            true => Val(GENCTRL0_A::GCLK8),
            i => Res(i),
        }
    }
    // More generated code here
}
```

burrbull, 2019-08-30:
see line 5816 in svd
incorrect bitwidth

srenauld, 2019-09-01:
correct - that's where the error was and all those fields should have been separate. I fixed that and now I'm one step further down the rabbit hole!

closed 2019-09-01 by srenauld