rust-embedded/svd2rust #386: `async` is a reserved keyword

Geobert, 2019-09-23:
It's easily fixed by search/replace whole word only and case sensitive `async` to `r#async` but I though it would be good to report this.

encountered when running `svd2rust` on http://www.keil.com/dd2/siliconlaboratories/efr32fg1p133f256gm48/

therealprof, 2019-09-23:
We'd be happy to take a PR... ;)

Geobert, 2019-09-23:
No promise (cobalt + rust-analyzer = huge list of things to be done) but I'll try as this seems to be a simple bug.
Just to be sure: shall I escape `async` or shall I name it something else?

adamgreig, 2019-09-23:
What version of svd2rust are you using? `async` is already listed as a reserved keyword in the sanitizer [here](https://github.com/rust-embedded/svd2rust/blob/master/src/util.rs#L71) so it shouldn't be getting output unescaped in the generated files.

Geobert, 2019-09-23:
I'm using `0.16.1`

martin31821, 2020-01-09:
FWIW, I see the same behavior on 0.17.0

burrbull, 2020-01-11:
Can you show SVD-source (or part of it with `async`) you try to convert?

jacobrosenthal, 2020-01-11:
Our efm32 
https://github.com/em32-rs/efm32hg-pac/blob/master/svd/EFM32HG309F64.svd

had to patch around this
https://github.com/em32-rs/efm32hg-pac/blob/master/devices/EFM32HG309F64.yaml#L5

at the very least 16 and 17 generate this

```
    #[doc = "Bit 28 - Asynchronous reflex"]
    #[inline(always)]
    pub fn async(&self) -> ASYNC_R {
        ASYNC_R::new(((self.bits >> 28) & 0x01) != 0)
    }
```
which when used give the expected
```
error: expected identifier, found keyword `async`
   --> /Users/jacobrosenthal/Downloads/efm32hg-pac/src/prs/ch0_ctrl.rs:384:12
    |
384 |     pub fn async(&self) -> ASYNC_R {
    |            ^^^^^ expected identifier, found keyword
    |
help: you can escape reserved keywords to use them as identifiers
    |
384 |     pub fn r#async(&self) -> ASYNC_R {
    |            ^^^^^^^

error: expected identifier, found keyword `async`
   --> /Users/jacobrosenthal/Downloads/efm32hg-pac/src/prs/ch0_ctrl.rs:406:12
    |
406 |     pub fn async(&mut self) -> ASYNC_W {
    |            ^^^^^ expected identifier, found keyword
    |
help: you can escape reserved keywords to use them as identifiers
    |
406 |     pub fn r#async(&mut self) -> ASYNC_W {
    |            ^^^^^^^
```

commit ee8a8a78 (2020-01-11): Merge #428

428: fix async sanitizing r=therealprof a=burrbull

Fix #386
PR #428 (2020-01-11): fix async sanitizing
Fix #386
closed 2020-01-11 by bors[bot]