rust-embedded/svd2rust #391: version 0.16.1 does not generate the library with all functions

heshananupama, 2019-10-15:
When I use the `svd2rust` version 0.12.1 and build the peripheral access crate for `stm32f301` the size of the `rlib` file is around 35mb. When I use the `objdump` to disassemble all the `rlib` file, it shows me all the functions and all the bit sequences in a proper manner.

However when I repeat the same process with version 0.16.1 the `rlib` file will have around 17mb and when I try to disassemble using `objdump` it only shows very few functions(less than 1% from the previous). I don't know what is the issue with the new version and hoping to use the older version as it works correctly. However, I would like to know the exact reason behind for this.

therealprof, 2019-10-15:
>  I don't know what is the issue with the new version and hoping to use the older version as it works correctly.

I'd presume that's due to inlining.

> as it works correctly.

Define "correctly". Having less useless code generated and smaller file sizes/faster build time is a feature, not a bug.

heshananupama, 2019-10-16:
@therealprof Yeah that's true. But using objdump of the `rlib`  generated from newer version does not give me all the functions as it does with the older version of `svd2rust` Using `objump -d stm32f30.rlib > functions.txt` would give me a txt file with only 100kb size with only few functions. As in the case of library generated from older version gives me a text file of 18mb with the disassembly code of all functions. I want to walk with disassembly code for research purpose and using newer version will not give me complete disassembly of the library

heshananupama, 2019-10-16:
Reduction in size is better but I am confused why `objdump` give two results. [A link to the 2 text file versions](https://drive.google.com/drive/folders/1OtJk3KNAcur2pFmMQcL8RXTZGPWnqkB2?usp=sharing)

therealprof, 2019-10-16:
Actually I suspect this might have something to do with LTO. In that case you wouldn't see any assembly due to the result being stored as LLVM IR. Can you try disabling LTO?

heshananupama, 2019-10-17:
Nope. I used the dev profile instead of release mode. I think LTO is off by default in dev mode. However, I tried explicitly stating it in the TOML file but it was the same.

therealprof, 2019-10-17:
TBH I have no idea what the contents of the rlib is even supposed to be. I guess some research is needed about the right approach to analyse this.

jonas-schievink, 2020-02-09:
rlibs are not object files, you cannot rely on them containing any symbols. In fact, there are plans to move all code generation into binary crates, leaving rlibs completely empty except metadata.

burrbull, 2021-04-17:
Is there any changes?