rust-embedded/svd2rust #392: use svd2rust from build.rs

stefanhoelzl, 2019-10-26:
I am trying to integrate svd2rust into my build.rs script and followed the [cargo book](https://doc.rust-lang.org/cargo/reference/build-scripts.html#case-study-code-generation)

I write the output of svd2rust into `$OUT_DIR/gen_lib.rs` and my `lib.rs` looks like the following:
```rust
include!(concat!(env!("OUT_DIR"), "/gen_lib.rs"));
```
but this doesn't compile.

Just having the following content in `gen_lib.rs`
```rust
#![no_std]
```
leads to the following error:
```bash
error: an inner attribute is not permitted in this context
 --> /Users/stefan/Development/k64/demo/target/debug/build/demo-77105b19538eb740/out/lib.rs:1:1
  |
1 | #![no_std]
  | ^^^^^^^^^^
  |
  = note: inner attributes, like `#![no_std]`, annotate the item enclosing them, and are usually found at the beginning of source files. Outer attributes, like `#[test]`, annotate the item following them.

error: expected item after attributes
 --> /Users/stefan/Development/k64/demo/target/debug/build/demo-77105b19538eb740/out/lib.rs:1:10
  |
1 | #![no_std]
  |          ^

error: aborting due to 2 previous errors
```

this is also one of severals errors I get when I try to use the `gen_lib.rs` generated by svd2rust.

Is there any way to get this working?

jonas-schievink, 2020-02-09:
I don't think svd2rust is meant to be used in this way. Doing this requires all users of the PAC to build svd2rust and sit through it processing the SVD (which can take some time), in addition to the already long compile times of the PACs themselves.

Emilgardis, 2020-02-10:
This is now possible since we provide svd2rust as a library. To use in build.rs use the supplied generate function to grab svd2rust output.

closed 2020-02-10 by Emilgardis