rust-embedded/svd2rust #404: Document how to use interrupts on RISC-V

xobs, 2019-12-28:
The implementation for interrupts generates a macro `interrupt!()` that takes the name of an interrupt, a closure, and optionally a struct of locals.

There are a few things that are unclear from the generated documentation:

1. Does the name of the ISR really matter?  This seems like an ARM-ism.
2. How does this handler actually get called?  The `riscv-rt` core calls a function called `trap_handler`, which gets set by default to `default_trap_handler()` which is a no-op.  Should the crate generated by `svd2rust` generate the `trap_handler()` to call the appropriate ISR, or should the `-rt` crate do so?

burrbull, 2019-12-28:
cc @Disasm

Disasm, 2019-12-28:
At the moment this part of interrupt handling is not implemented at all. The only thing used is `trap_handler`. You can override the default trap handler to handle everything: exceptions, interrupts and IRQs.

How it should be implemented? Well, I don't have a solution yet. Maybe there should be a function, like `irq_handler` with a default implementation in `riscv-rt`. Overridden version of this function will be declared in svd2rust-generated crate and will call one of the `<IRQ_NAME>()` handlers.

luojia65, 2020-03-01:
Some chips may have its own `irq_handler` routime (e.g. to speed up tail-biting interruption using special CSRs where a `csrrw` actually jumps `pc` into another location), so if there are the default handlers for IRQ and/or trap they should be weak functions allowing downstream crates to alter.

Disasm, 2020-03-01:
Soon: https://github.com/rust-embedded/riscv-rt/pull/42
At the moment, the plan is the following:

- Release riscv-rt with interrupt support
- Generate IRQ handler names, handler array and weak defaults with svd2rust
- Add platform-dependent implementations of the `MachineExternal()` handler that will call these IRQ handlers
- Add macro (or proc-macro) to define custom IRQ handlers

burrbull, 2022-07-19:
Should be fixed, so closing.