rust-embedded/svd2rust #433: Hide peripherals under a feature gate each one.

jeandudey, 2020-02-23:
Compiling a complete crate generated using `svd2rust` results in a big compilation time (at least for my machine), mainly because of all the code that needs to be compiled. Adding a feature gate to each peripherals results in better compilation times.

For example one would use a PAC crate like this by specifying the features on demand:

```toml
[dependencies.mymcu-pac]
version = "0.1"
features = ["uart", "gpio"]
```

A flag to `svd2rust` could be added to generate code with features on peripherals like `--use-features` (bad name, but gets the point) so I doesn't break existing code and is enabled explicitly.

burrbull, 2020-02-24:
This idea is not new.

It was even implemented. See #236, #254 (partially rebased https://github.com/burrbull/svd2rust/tree/gates)

But some people against this approach
https://github.com/rust-embedded/svd2rust/pull/254#issuecomment-437713270

jeandudey, 2020-02-24:
Quoting from https://github.com/rust-embedded/svd2rust/pull/254#issuecomment-437713270

>As raised on IRC I am not happy with the approach due to the usability complications this causes for the mere benefit of some compile time gain.

It's not actually some compile time gain, it's also higher memory usage of the `rustc` compiler, not every computer has a lots of resources to just build a PAC (which limits the reach of this project) and also limits other areas like using `svd2rust` generated crates on resource constrained systems, see the case of using a Raspberry Pi 3 Model B, has 1 GB of RAM (my computer crashed building at least 3 different crates at different times, I have 4 GB of RAM, hit swap partition of course). While you may argue that it's a problem of the Rust compiler per se (and it is), it doesn't mean that downstream crate users need to be limited in where and how to compile these crates.

>In my opinion we should try to simplify the use of embedded rust crates and increase the portability instead of pinning everything down to the last bit requiring changes all over the map even if just a small detail in the hardware changes.

Features aren't that complex at all, and can be enabled on a by-default basis, and _those who do not want all the peripherals_ can disable them and enable only the needed.

adamgreig, 2020-02-24:
This is sort of well trodden territory but in principle the idea could make sense. It would definitely help reduce compile time and memory load.

One use case I'd be personally worried about is in for example stm32-rs where we combine many svd2rust-generated PACs into a single crate, with a feature to select which device to use. Right now svd2rust does not generate a `Cargo.toml` file, but presumably if it's generating hundreds of features it would need to at least generate the `Cargo.toml` entries for those features. However, in stm32-rs the `Cargo.toml` already has autogenerated features per device, so somehow these features would need to be merged together from each device and with the per-device features. Not the end of the world but the implementation details could make this very easy or very hard.

therealprof, 2020-02-24:
I totally see your but but I totally disagree with your conclusion:

> Features aren't that complex at all, and can be enabled on a by-default basis, and those who do not want all the peripherals can disable them and enable only the needed.

Features are a pain in the neck and create a lot of usability and composability problems now and in the future. Also it is absolutely not clear where you would derive the features from.

I'm not saying I cannot be convinced otherwise but without some elaborate proof-of-concept that at least demonstrates that it's not a total nightmare **and** an automisable process, this seems rather unlikely.

I'd say the energy is better invested in figuring out a way to make to the improve the current approach and/or the compiler. For everything else there's also `stm32-ral` which is a whole lot more lightweight and also binary efficient in dev mode.

duskmoon314, 2022-08-03:
In [v0.25.0](https://github.com/rust-embedded/svd2rust/blob/master/CHANGELOG.md#v0250---2022-08-02), we introduce `feature_group` and `feature_peripheral` to control which peripheral can be seen. (All peripherals are enabled by default)

I think this issue has been solved. 😃

closed 2022-08-03 by jeandudey