rust-embedded/svd2rust #434: Support `WriteConstraint` on a Register Level

cr1901, 2020-04-11:
The CMSIS SVD Specification allows `WriteConstraint` fields to be applied to a `Register` type, and _not_ just its fields:

```
  <xs:complexType name="registerType">
    <xs:sequence>
...
     <!-- writeConstraint specifies the subset of allowed write values -->
      <xs:element name="writeConstraint" type="writeConstraintType" minOccurs="0"/>
...
    </xs:sequence>
    <xs:attribute name="derivedFrom" type="dimableIdentifierType" use="optional"/>
  </xs:complexType>
```

Right now, `svd2rust` has no provisions for providing a safe `bits()` function which accesses the entire register (and not just a field), because the generics `bits()` function for writing a register is- _correctly_ [marked](https://github.com/rust-embedded/svd2rust/blob/master/src/generate/generic.rs#L244-L251) as unsafe!

To avoid having application writers have to place unsafe code in their application, I propose some standardized function to _safely_ access all the bits of a register, as a unit, when `svd2rust` can prove that no invalid values can be written into a register. 

`svd2rust` could either prove this info by reading the `WriteConstraint` field of a `Register` type, or keeping a running tally of which bits are stated to be valid in children (`fieldType`) of the `fieldsType` sequence.

This is useful for RMW GPIO registers (direction, value, etc) and setting timer values for countdown all at once, without having to chain proxies together for each individual bit.

burrbull, 2022-10-22:
Related PR: #666

burrbull, 2022-11-07:
reopen if not fixed yet

closed 2022-11-07 by burrbull