rust-embedded/svd2rust #437: Crates generated with svd2rust 0.17.0 cause warnings on stable

hannobraun, 2020-04-22:
Here's what I'm getting when compiling with 1.42.0:
```
warning: lint `legacy_directory_ownership` has been removed: `converted into hard error, see https://github.com/rust-lang/rust/issues/37872`
 --> src/lib.rs:5:9
  |
5 | #![deny(legacy_directory_ownership)]
  |         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(renamed_and_removed_lints)]` on by default

warning: lint `plugin_as_library` has been removed: `plugins have been deprecated and retired`
  --> src/lib.rs:12:9
   |
12 | #![deny(plugin_as_library)]
   |         ^^^^^^^^^^^^^^^^^

warning: lint `safe_extern_statics` has been removed: `converted into hard error, see https://github.com/rust-lang/rust/issues/36247`
  --> src/lib.rs:14:9
   |
14 | #![deny(safe_extern_statics)]
   |         ^^^^^^^^^^^^^^^^^^^

warning: unknown lint: `unions_with_drop_fields`
  --> src/lib.rs:16:9
   |
16 | #![deny(unions_with_drop_fields)]
   |         ^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(unknown_lints)]` on by default
```

therealprof, 2020-04-22:
Fixed by #421, duplicate of #415 

Probably should a release a some point...

closed 2020-04-22 by therealprof

hannobraun, 2020-04-22:
Oh, sorry, I only checked the open issues for duplicates. Didn't think to check the closed ones.

therealprof, 2020-04-22:
No worries, thanks for the report.