rust-embedded/svd2rust #450: Cannot generate code with Renesas RA controller

jolo1581, 2020-06-17:
Hi,

I am using the SVD file from Renesas found here:
[https://github.com/renesas/fsp/tree/master/ra/fsp/src/bsp/cmsis/Device/RENESAS/SVD](url)

I installed svd2rust by cargo command.
Now when I use `svd2rust -i RA.svd` I got only errors:

svd2rust -i RA.svd             
thread 'main' panicked at 'assertion failed: info.name.contains("%s")', /home/jan/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.9.0/src/svd/register.rs:43:13
stack backtrace:
   0:     0x560a97a85e14 - backtrace::backtrace::libunwind::trace::hc1c4a1d8ad423b97
                               at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
   1:     0x560a97a85e14 - backtrace::backtrace::trace_unsynchronized::h82274781060cb056
                               at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
   2:     0x560a97a85e14 - std::sys_common::backtrace::_print_fmt::h2a45d89b653a4da8
                               at src/libstd/sys_common/backtrace.rs:78
   3:     0x560a97a85e14 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h41a0a93ab85e6aa1
                               at src/libstd/sys_common/backtrace.rs:59
   4:     0x560a97aac92c - core::fmt::write::hdaea18585065a96d
                               at src/libcore/fmt/mod.rs:1069
   5:     0x560a97a83b53 - std::io::Write::write_fmt::h0cea70c809005252
                               at src/libstd/io/mod.rs:1504
   6:     0x560a97a886b5 - std::sys_common::backtrace::_print::hd95f9978cc145ca4
                               at src/libstd/sys_common/backtrace.rs:62
   7:     0x560a97a886b5 - std::sys_common::backtrace::print::hfb25ca2291be47d0
                               at src/libstd/sys_common/backtrace.rs:49
   8:     0x560a97a886b5 - std::panicking::default_hook::{{closure}}::h44f76cee5dc8591c
                               at src/libstd/panicking.rs:198
   9:     0x560a97a883f2 - std::panicking::default_hook::h198e1a712910f1e6
                               at src/libstd/panicking.rs:218
  10:     0x560a97a88d12 - std::panicking::rust_panic_with_hook::hc0b4730bb8013f9d
                               at src/libstd/panicking.rs:511
  11:     0x560a97a123ae - std::panicking::begin_panic::hf51d9e42249bf3e2
  12:     0x560a97a0a38d - <svd_parser::svd::register::Register as svd_parser::parse::Parse>::parse::h61a4471fffd977f8
  13:     0x560a97a0a495 - <svd_parser::svd::registercluster::RegisterCluster as svd_parser::parse::Parse>::parse::hfc103e587b57c3a9
  14:     0x560a97a0fa0b - <core::iter::adapters::ResultShunt<I,E> as core::iter::traits::iterator::Iterator>::next::h80825df8f6575409
  15:     0x560a97a0dd51 - <alloc::vec::Vec<T> as alloc::vec::SpecExtend<T,I>>::from_iter::h590a698023dc0c65
  16:     0x560a97a10607 - core::iter::adapters::process_results::h8238f7e61ee600b3
  17:     0x560a979fe757 - <svd_parser::svd::clusterinfo::ClusterInfo as svd_parser::parse::Parse>::parse::ha4da8b23272e051a
  18:     0x560a979f6faa - <svd_parser::svd::cluster::Cluster as svd_parser::parse::Parse>::parse::hb51672cd03d7801c
  19:     0x560a97a0a5ef - <svd_parser::svd::registercluster::RegisterCluster as svd_parser::parse::Parse>::parse::hfc103e587b57c3a9
  20:     0x560a97a0fb62 - <core::iter::adapters::ResultShunt<I,E> as core::iter::traits::iterator::Iterator>::next::hf3733e86292e3f52
  21:     0x560a97a0def1 - <alloc::vec::Vec<T> as alloc::vec::SpecExtend<T,I>>::from_iter::h888d0d22943c4375
  22:     0x560a97a103a7 - core::iter::adapters::process_results::h1d041bfb76276ec1
  23:     0x560a97a0083e - <svd_parser::svd::peripheral::Peripheral as svd_parser::parse::Parse>::parse::h5068b2d95ce8454d
  24:     0x560a97a0e6c8 - <core::iter::adapters::ResultShunt<I,E> as core::iter::traits::iterator::Iterator>::next::h01e498058b701949
  25:     0x560a97a0da7a - <alloc::vec::Vec<T> as alloc::vec::SpecExtend<T,I>>::from_iter::h45208e3bac7b1db9
  26:     0x560a97a104d7 - core::iter::adapters::process_results::h65d29583847104ab
  27:     0x560a979fa9d3 - <svd_parser::svd::device::Device as svd_parser::parse::Parse>::parse::h600639b2a002e3ca
  28:     0x560a979f79a9 - svd_parser::parse::hb5d4496179ada47a
  29:     0x560a977c0864 - svd2rust::main::h45c70d52bcb61431
  30:     0x560a977bbb73 - std::rt::lang_start::{{closure}}::h621867c08da5c03d
  31:     0x560a97a89228 - std::rt::lang_start_internal::{{closure}}::h7a212202ff76034b
                               at src/libstd/rt.rs:52
  32:     0x560a97a89228 - std::panicking::try::do_call::h6d214a73427d759b
                               at src/libstd/panicking.rs:331
  33:     0x560a97a89228 - std::panicking::try::hc078f0e11721d1cb
                               at src/libstd/panicking.rs:274
  34:     0x560a97a89228 - std::panic::catch_unwind::hb31c05be30625612
                               at src/libstd/panic.rs:394
  35:     0x560a97a89228 - std::rt::lang_start_internal::hcf7fb98a775d5af0
                               at src/libstd/rt.rs:51
  36:     0x560a977c2632 - main
  37:     0x7f43e297c002 - __libc_start_main
  38:     0x560a9779f17e - _start
  39:                0x0 - <unknown>

Can somebody help me please :-)

jwinarske, 2020-06-18:
You might want to validate the XML, then validate it against the (xsd)[https://raw.githubusercontent.com/ARM-software/CMSIS_5/develop/CMSIS/Utilities/CMSIS-SVD.xsd].  Once that all passes, then try it with svd2rust.

jolo1581, 2020-06-19:
Seems that I am not the only user with this problem.

https://renesasrulz.com/ra/f/forum/16396/ra-svd-on-github-fails-xsd-validation

jolo1581, 2020-06-21:
What can I do next?
jan@elite:~/Projects $ xmllint --schema CMSIS-SVD.xsd RA.svd --noout
RA.svd:270: element name: Schemas validity error : Element 'name': This element is not expected.
RA.svd:22557: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:22652: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:35567: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:35573: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:35579: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:49059: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:49069: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:49079: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:50536: element cluster: Schemas validity error : Element 'cluster': This element is not expected. Expected is one of ( alternateCluster, headerStructName, addressOffset ).
RA.svd:56441: element description: Schemas validity error : Element 'description': [facet 'minLength'] The value has a length of '0'; this underruns the allowed minimum length of '1'.
RA.svd:70088: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70099: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70110: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70121: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70132: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70142: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70153: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70164: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70175: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70186: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70196: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70207: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70218: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70229: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd:70240: element fields: Schemas validity error : Element 'fields': Missing child element(s). Expected is ( field ).
RA.svd fails to validate

jwinarske, 2020-06-22:
@jolo1581 you need to raise the issue with Renesas, or fix the file yourself.

jolo1581, 2020-06-24:
I solved the problem.
In IAR embedded workbench is a svd file that is valid for RA2

closed 2020-06-24 by jolo1581