rust-embedded/svd2rust #453: Update for `bare-metal` v1.0.0

ForsakenHarmony, 2020-07-02:
https://github.com/rust-embedded/bare-metal/blob/master/CHANGELOG.md#v100---2020-06-23

Same org, so maybe you're already doing it internally, just thought I'd submit this issue because I didn't find anything

therealprof, 2020-07-05:
Just for reference, this manifests as:

```
error[E0405]: cannot find trait `Nr` in crate `bare_metal`
  --> src/lib.rs:68:40
   |
68 | USB = 31 , } unsafe impl bare_metal :: Nr for Interrupt { # [ inline ( always ) ]
   |                                        ^^ not found in `bare_metal`
   |
help: possible candidate is found in another module, you can import it into scope
   |
22 | extern crate cortex_m_rt ; extern crate bare_metal ; extern crate vcell ; use cortex_m::interrupt::Nr;
   |                                                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0405`.
error: could not compile `stm32f0x-test`.
```

CC rust-embedded/bare-metal#32

burrbull, 2021-04-17:
Close by #473

closed 2021-04-17 by burrbull