rust-embedded/svd2rust #459: Awkward to write code generic over registers

couchand, 2020-08-05:
I am writing code that is generic over registers, and I'm getting a bit frustrated by the awkward bounds that need to be applied.

The ideal case would look something like this:

```rust
fn read_bit<U: Copy, Register: Readable>(reg: &Reg<U, Register>) -> U {
    // ...
}
```

Unfortunately, the `Readable` and `Writable` traits are impl'd on the `Reg` struct typedef itself, not on the inner type.  Also unfortunately, the traits don't provide any functionality directly, so I can't rely completely on the traits and forget about the `Reg` struct.  This means the bounds need to be written in the much more awkward form:

```rust
fn read_bit<U: Copy, Register>(reg: &Reg<U, Register>) -> U
where
    Reg<U, Register>: Readable,
{
    // ...
}
```

In my case, I'd like to apply the bounds to associated types in a trait.  Though I can add the relevant where bounds on `Reg<...>` to the trait definition, they don't "propagate" to uses of that trait, meaning I need to repeat them on any block that uses the trait.

Ideally it would look something like:

```rust
trait Port {
    type U: Sized + Copy + BitAnd + BitOr + ...;
    type DDR: Readable + Writable;
    type Data: Readable + Writable;

    fn ddr() -> Reg<Self::U, Self::DDR>;
    fn data() -> Reg<Self::U, Self::Data>;
}

impl<P: Port> Component<P> {
    fn configure() {
        P::ddr().modify(|r, w| {
            // ...
        });
    }

    fn do_something() {
        P::data().modify(|r, w| {
            // ...
        });
    }
}
```

But, given the above constraints, I actually need to write:

```rust
trait Port { // I could put a where clause here but it really wouldn't help
    type U: Sized + Copy + BitAnd + BitOr + ...;
    type DDR;
    type Data;

    fn ddr() -> Reg<Self::U, Self::DDR>;
    fn data() -> Reg<Self::U, Self::Data>;
}

impl<P: Port> Component<P>
where
    Reg<P::U, P::DDR>: Readable + Writable
{
    fn configure() {
        P::ddr().modify(|r, w| {
            // ...
        });
    }
}

impl<P: Port> Component<P>
where
    Reg<P::U, P::Data>: Readable + Writable
{
    fn do_something() {
        P::data().modify(|r, w| {
            // ...
        });
    }
}
```

I believe that this situation could be resolved by generating the `impl Readable`/`impl Writable` for the hidden inner register type rather than directly on the outer register type alias.  For documentation purposes, there could be a blanket impl on `Reg`, like:

```rust
impl<U, REG: Readable> Readable for Reg<U, REG> {}
impl<U, REG: Writable> Writable for Reg<U, REG> {}
```

Which I believe would be allowed due to coherence.

If you are open to these changes, I would be willing to work on a PR implementing them.  Or if you have suggestions for another strategy to make this usage more ergonomic, I'm all ears.

I spent a while looking around for related discussion and found none.  Feel free to point me to any existing documentation or prior discussion on this topic.

Thanks for all your hard work on this!

therealprof, 2020-08-05:
Hi there. 👋

Thanks for bringing this up. The main motivation behind the introduction of those traits was code and repetition reduction, but there's no reason why we couldn't improve the flexibility and usability at the same time.

I absolutely think that we should work on the interfaces (and also on the generated documentation which seems to have suffered a bit) and your suggestion makes perfect sense to me, although I have to admit I don't quite see where you're going with this. 😅

Please feel free to work on this idea, if you'd like we can also nominate this issue for discussion in the next WG meeting (every Tuesday 8:00 pm Berlin time on Matrix/Element, open to everyone) and talk about it.

ryankurte, 2020-08-08:
excellent progress! bors appears to have been a little over eager on closing this

therealprof, 2020-08-08:
Well, since the author instructed bors to close this issue, I guess we're good. ;)

Also CC #465 and #466 for being related.

commit 53569906 (2020-08-07): Merge #464

464: Register spec & traits r=therealprof a=couchand

Part 1 of #463.

- Publishes the register spec zero-sized type and moves all relevant register traits to that struct, which fixes #459 and improves generated documentation.
- Removes the extra type parameter on `Reg`, making the register spec the sole authority on the shape of the register (but not its address).

One thing that's changed since the previous PR, this renames the spec struct so that it won't cause a conflict on NXP, Freescale, and possibly other architectures.  I've gone with `register::REGISTER_SPEC` since that se [... cut]
PR #464 (2020-08-07): Register spec & traits
Part 1 of #463.

- Publishes the register spec zero-sized type and moves all relevant register traits to that struct, which fixes #459 and improves generated documentation.
- Removes the extra type parameter on `Reg`, making the register spec the sole authority on the shape of the register (but not its address).

One thing that's changed since the previous PR, this renames the spec struct so that it won't cause a conflict on NXP, Freescale, and possibly other architectures.  I've gone with `register::REGISTER_SPEC` since that seems reasonably clear and I can't think of h [... cut]
closed 2020-08-08 by therealprof