rust-embedded/svd2rust #474: Enumerated values derivedFrom regression

MabezDev, 2020-09-20:
DerivedFrom registers are not generating getters for the EnumeratedVariants.

esp32-0.8.0 (generated using svd2rust v0.17.0 (714ed98 2020-06-09)):
```rust
// initial register
#[doc = "Reader of field `WDT_STG0`"]
pub type WDT_STG0_R = crate::R<u8, WDT_STG0_A>;
impl WDT_STG0_R {
    #[doc = r"Get enumerated values variant"]
    #[inline(always)]
    pub fn variant(&self) -> crate::Variant<u8, WDT_STG0_A> {
        use crate::Variant::*;
        match self.bits {
            0 => Val(WDT_STG0_A::DISABLE),
            1 => Val(WDT_STG0_A::INTERRUPT),
            2 => Val(WDT_STG0_A::RESETCPU),
            3 => Val(WDT_STG0_A::RESETSYSTEM),
            4 => Val(WDT_STG0_A::RESETRTC),
            i => Res(i),
        }
    }

// ..

// derived from:

#[doc = ""]
pub type WDT_STG1_A = WDT_STG0_A;
#[doc = "Reader of field `WDT_STG1`"]
pub type WDT_STG1_R = crate::R<u8, WDT_STG1_A>;
```

Essentially `crate::R<u8, WDT_STG0_A>` has all variant method impls, and the derivedFrom versions are simply type alias to that.

However in esp32-0.9.0 (generated using svd2rust v0.17.0 (1a2bf65 2020-09-20)) : 

```rust
#[doc = "Field `WDT_STG0` reader - "]
pub struct WDT_STG0_R(crate::FieldReader<u8, WDT_STG0_A>);
impl WDT_STG0_R {
    pub(crate) fn new(bits: u8) -> Self {
        WDT_STG0_R(crate::FieldReader::new(bits))
    }
    #[doc = r"Get enumerated values variant"]
    #[inline(always)]
    pub fn variant(&self) -> crate::Variant<u8, WDT_STG0_A> {
        use crate::Variant::*;
        match self.bits {
            0 => Val(WDT_STG0_A::DISABLE),
            1 => Val(WDT_STG0_A::INTERRUPT),
            2 => Val(WDT_STG0_A::RESETCPU),
            3 => Val(WDT_STG0_A::RESETSYSTEM),
            4 => Val(WDT_STG0_A::RESETRTC),
            i => Res(i),
        }
    }
   // .. other impls omitted fro brevity

  // derived from:

#[doc = ""]
pub type WDT_STG1_A = WDT_STG0_A;
#[doc = "Field `WDT_STG1` reader - "]
pub struct WDT_STG1_R(crate::FieldReader<u8, WDT_STG1_A>);
impl WDT_STG1_R {
    pub(crate) fn new(bits: u8) -> Self {
        WDT_STG1_R(crate::FieldReader::new(bits))
    }
}
```

Each derivedFrom is a new type, but as you can see it doesn't have the variant methods generated for it.

It seems to me like the methods should be generated for all derived from structs (duplicating code :() or go back to the trait alias method?

couchand, 2020-09-23:
@burrbull I suspect this was me, feel free to assign me, I'll take a look tomorrow.

couchand, 2020-09-23:
@MabezDev to clarify the rationale for the change: putting all the various methods as `impl`s on the one `R` struct made the documentation basically impenetrable.  I'd rather not go back to making everything an alias for `R`.

However, in this case, it would make a lot of sense to have just `WDT_STG1_R` be a type alias, something like:

```rust
#[doc = "Reader of field `WDT_STG1`"]
pub type WDT_STG1_R = WDT_STG0_R;
```

burrbull, 2020-09-23:
Yes, it is a bug.
It must be just type alias as it was before.

closed 2020-09-23 by bors[bot]

MabezDev, 2020-09-23:
That works perfectly! Thanks for the speedy fix @couchand

couchand, 2020-09-23:
Always happy to fix the things that I broke, @MabezDev!

commit 5700351d (2020-09-23): derivedFrom FieldReaders should always be a type alias for the base FieldReader

Fixes #474
commit 29129c0e (2020-09-23): Merge #476

476: Fix derivedFrom enumerated values r=therealprof a=couchand

derivedFrom FieldReaders should always be a type alias for the base FieldReader, rather than each their own newtype.

~I believe this fixes #474, however I'm having trouble getting a working development environment set up for the `esp32` project.~

~@MabezDev, would you mind giving this a spin and posting a diff of the generated code?~

Figured out my issue, the generated code looks good, e.g.:

```diff
diff --git a/src/rtccntl/wdtconfig0.rs b/src/rtccntl/wdtconfig0.rs
index 6851bff..089e4e1 100644
--- a/src/rtccntl/w [... cut]
PR #476 (2020-09-23): Fix derivedFrom enumerated values
derivedFrom FieldReaders should always be a type alias for the base FieldReader, rather than each their own newtype.

~I believe this fixes #474, however I'm having trouble getting a working development environment set up for the `esp32` project.~

~@MabezDev, would you mind giving this a spin and posting a diff of the generated code?~

Figured out my issue, the generated code looks good, e.g.:

```diff
diff --git a/src/rtccntl/wdtconfig0.rs b/src/rtccntl/wdtconfig0.rs
index 6851bff..089e4e1 100644
--- a/src/rtccntl/wdtconfig0.rs
+++ b/src/rtccntl/wdtconfig0.r [... cut]