rust-embedded/svd2rust #487: Generated code doesn't work

thejpster, 2021-01-24:
I built a PAC using svd2rust 0.17 and it didn't compile (https://github.com/rp-rs/rp2040-pac).

I had to:

* Remove a bunch of lints that rustc 1.49 didn't like
* `Replace bare_metal::Nr` with `cortex_m::interrupt::Nr`

I also really struggled with the docs, which in the SVD are full of `\n` character pairs. It would really help if svd2rust could process out these character pairs and split the docs into multiple lines.

thejpster, 2021-01-24:
I can confirm that head works fine. See https://github.com/rp-rs/rp2040-pac/tree/use_svd_0.18alpha

therealprof, 2021-01-30:
We were just discussing this on the `rp-rs` channel and @adamgreig mentioned that the `\n` are not really supposed to be there. We might try to filter them out in the parser if it's not too painful/expensive.

thejpster, 2021-01-30:
I tried to do it in the generated output with sed and it was not easy at all (because you use #[doc=""] not ///). Filtering the string before it's rendered as docs would be best.

At least they didn't use Doxygen syntax in the comments 😂

therealprof, 2021-01-31:
There's unfortunately no way to generate `///` doc strings, it's a long pet peeve of mine https://github.com/alexcrichton/proc-macro2/issues/205

burrbull, 2021-05-31:
Is this still actual?

closed 2022-07-19 by burrbull