rust-embedded/svd2rust #492: Support arrays of peripherals

daym, 2021-02-21:
It would be nice if svd2rust also supported peripheral arrays of SVD 1.3.

See https://arm-software.github.io/CMSIS_5/SVD/html/elem_peripherals.html#elem_peripheral .

```
Starting version 1.3 of the SVD specification, arrays of peripherals can be specified. The single peripheral description gets duplicated automatically into an array. The number of array elements is specified by the <dim> element. To define arrays, the <name> needs the format myPeripheral[%s]. The tag <dimIncrement> specifies the address offset between two peripherals. To create copies of a peripheral using different names, you must use the derivedFrom attribute.
```

duskmoon314, 2022-04-26:
@daym This feature has been implemented and merged. 😄

Emilgardis, 2022-04-26:
Fixed in #592

Emilgardis, 2022-04-26:
For your information @duskmoon314 you can auto-close issues with the following syntax: https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

Very useful :D

closed 2022-04-26 by Emilgardis