rust-embedded/svd2rust #494: Indirect access registers

daym, 2021-03-10:
It would be nice to be able to use svd2rust for SVDs that describe indirect accesses via a data port register. (for example the infamous pci index and pci data register)

I think it would be easy to extend svd2rust to do that, but I wanted to ask first whether 

(1) There will be unforseen hurdles

(2) Whether there is interest in that

adamgreig, 2021-03-10:
What sort of thing did you have in mind? I don't think the SVD specification has any support for indicating data port registers, so I don't know how you'd plan to tell svd2rust to generate whatever special output you needed.

It seems like this sort of thing would usually be handled by a higher level (a HAL) which would know how to access the data port and could present a more convenient API to the user.

daym, 2021-03-10:
In https://github.com/daym/amd-register-interface-extractor we currently emit one svd file for each access method, including one for direct access. So we'd know that we need an indirect access by virtue of this svd2rust invocation being with a svd file with all indirect accesses.

How the indirect access is done would need to be provided by the user.

If this is not really something svd2rust is supposed to do, we can also write our own tool, I guess.

For an overview, we have 24802 direct-access register instances, 2032 AMD-specific indirect access register instances, 6 IO-accessed register instances, 15 indirect IO-accessed register instances, 331 MSR-accessed register instances and 16140 indirect SMN-accesses register instances.

adamgreig, 2021-03-13:
That's a cool use case and I can see why you'd want to use svd2rust!

It sounds like the ideal situation would be if the sort of "user facing API generator" was separate so you could use it without the MMIO assumption. My feeling is that the svd2rust binary (at least the one we provide here) shouldn't include this sort of thing because it's completely out of scope - SVD files are explicitly only about MMIO, the format doesn't have anything to indicate an alternative indirect access, and indirect access to config/status registers tends to be a very rare feature in embedded microcontrollers.

There _is_ interest in re-designing parts of the generated code so that instead of memory-mapped structs, we use only raw pointers and ptr::read_volatile/ptr::write_volatile, because of a long-standing issue where rustc can insert reads wherever it wants to references, which we need to avoid for MMIO. I could imagine that with such a change it might be possible to also have an alternative codegen that uses indirect access, but again we probably wouldn't want to include it in svd2rust-the-binary because it's so out of scope. Maybe it would be possible for you to build a binary that depends on svd2rust for almost all the generation, but then emits a custom indirect access method? I'm not sure exactly what this would look like, and perhaps once you've been able to prototype something on your side it would be clearer where it might tie in.