rust-embedded/svd2rust #504: More duplicate structs generated.

john-terrell, 2021-04-15:
I'm unsure I this is a dupe of #357.   The SVD [here](https://github.com/atsam-rs/atsam-pac/blob/john/sam4l_support/svd/ATSAM4LC2A.svd) creates duplicate structs.

error[E0428]: the name `CMR` is defined multiple times
   --> pac/atsam4lc2a/src/tc0.rs:151:1
    |
147 | pub type CMR = crate::Reg<cmr::CMR_SPEC>;
    | ----------------------------------------- previous definition of the type `CMR` here
...
151 | pub type CMR = crate::Reg<cmr::CMR_SPEC>;
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `CMR` redefined here
    |
    = note: `CMR` must be defined only once in the type namespace of this module

And the code does indeed contain duplicates:

#[doc = "Channel Control Register Channel"]
pub mod ccr;
**#[doc = "CMR register accessor: an alias for `Reg<CMR_SPEC>`"]
pub type CMR = crate::Reg<cmr::CMR_SPEC>;
#[doc = "Channel Mode Register Channel"]
pub mod cmr;**
**#[doc = "CMR register accessor: an alias for `Reg<CMR_SPEC>`"]
pub type CMR = crate::Reg<cmr::CMR_SPEC>;
#[doc = "Channel Mode Register Channel"]
pub mod cmr;**
#[doc = "CV register accessor: an alias for `Reg<CV_SPEC>`"]
pub type CV = crate::Reg<cv::CV_SPEC>;

burrbull, 2021-04-15:
Can you try different versions of `svd2rust` an investigate what version (commit) causes this?

burrbull, 2021-04-15:
I see such output when run:
```sh
[WARN  svd2rust::generate::peripheral] Found type name conflict with region Some("cmr"), renamed to Some("cmr0")
[WARN  svd2rust::generate::peripheral] Found type name conflict with region Some("cmr"), renamed to Some("cmr1")
[WARN  svd2rust::generate::peripheral] Found type name conflict with region Some("cmr"), renamed to Some("cmr2")
```

john-terrell, 2021-04-15:
I tried master and the head of the v0.18 branch.

burrbull, 2021-04-15:
CMR is twice described. Check your SVD more carefully. 
[image]

john-terrell, 2021-04-15:
I've fixed this by patching the SVD.   The SVD has duplicate register names with different <alternateGroup> values.

e.g.

<register>
          <name>CSR</name>
          <description>Channel Status Register</description>
          <alternateGroup>SPI_SLAVE_MODE</alternateGroup>
          <addressOffset>0x14</addressOffset>
          <size>32</size>

...

<register>
          <name>CSR</name>
          <description>Channel Status Register</description>
          <alternateGroup>USART_MODE</alternateGroup>
          <addressOffset>0x14</addressOffset>
          <size>32</size>
          <access>read-only</access>

burrbull, 2021-04-15:
`alternateGroup`'s are not supported yet in `svd2rust`. I see 2 ways to do this:
1. Add `ALTERNATE_GROUP_NAME_` prefix for such registers.
2. Put them in different modules and access them using `alternate_group_name::REGNAME`

burrbull, 2021-04-15:
I tried to implement first variant but looks like it works for all registers with `alternateGroup` in your file except `CMR`

burrbull, 2021-04-15:
Look #506. It should fix your issue.

burrbull, 2021-04-17:
Should be fixed now

closed 2021-04-17 by burrbull