rust-embedded/svd2rust #513: Question on alternateGroup

nickray, 2021-05-02:
https://github.com/rust-embedded/svd2rust/pull/506 introduced "alternateGroup support".

In the case of LPC55 (e.g. https://github.com/lpc55/lpc55-pac/blob/d53269c5e87ae5151f780203702b548584dd0610/svd/pack-13.0.0-LPC55S69_cm33_core0.xml) this completely changed the svd2rust API all over the place.

I'm not sure if this was intended/foreseen?

Also not sure if this is a bug of sorts in the SVD; the net effect is it makes a lot of names very long (which I'm tending towards reverting by simply sed-ing out all the alternateGroup lines).

burrbull, 2021-05-02:
Looking at SVD you cite as an example I see incorrect usage of `alternateGroup`.
For example `SCT0` peripheral.
`CAP14` and `MATCH14` both at address `0x138`. And both in one `CAP_MATCH` `alternateGroup`.
If `alternateGroup` is used they must be in different groups. All `CAPx` in `CAP` group and all `MATCHx` in `MATCH` group.
So LPC SVDs are bugged.

The only way to workaround this is to add flag that disables #506 for such files.

burrbull, 2021-05-08:
See #516

burrbull, 2021-05-08:
You can now use `--ignore_groups` for old behavior.