rust-embedded/svd2rust #514: Error has low context when parsing presumably invalid XML

hegza, 2021-05-06:
## Preconditions:
We have an experimental SVD generator that produces presumably incorrect CMSIS-SVD XML: file.svd.

## What I did
Compiled svd2rust from master branch in debug mode, then executed:

`RUST_LOG=trace ./svd2rust -i file.svd --log=trace --target=riscv`

with all combinations of `RUST_LOG` & `--log` for good measure. Additionally, same results with `svd2rust` from `cargo install svd2rust`.

## Output
> [ERROR svd2rust] expected `,`

## Expected output
More context. Like, possibly a line number or more debug/trace. Discussion on Rust Embedded at Matrix yields that likely source is svd-parser itself or the error interface to it. Example output:
> [TRACE svd2rust] parsing line 56
> [ERROR svd2rust] expected ',' on line 56

burrbull, 2021-05-08:
Related to https://github.com/rust-embedded/svd/issues/18

burrbull, 2022-07-19:
Parsing errors now much better.
Reopen if still actual.

closed 2022-07-19 by burrbull