rust-embedded/svd2rust #533: Crashing on RISC-V E74-MC Unmatched .svd

U007D, 2021-07-08:
When parsing https://github.com/sifive/freedom-e-sdk/blob/origin/dev/zongl/unmatched/bsp/sifive-hifive-unmatched/design.svd (attached as `sifive_hifive_unmatched.svd.zip`), I get the following (note the lack of backtrace):
```
sergibli@HephaestusMobi:~/Development/svd2rust
▶ RUST_BACKTRACE=1 svd2rust --target riscv -i ../bg/ada/sifive_hifive_unmatched_hal/sifive_hifive_unmatched.svd -o ../bg/ada/sifive_hifive_unmatched_hal/output/
fatal runtime error: failed to initiate panic, error 5
Abort trap: 6

sergibli@HephaestusMobi:~/Development/svd2rust
▶ svd2rust --version
svd2rust 0.19.0 ( )

sergibli@HephaestusMobi:~/Development/svd2rust
▶ rustc --version
rustc 1.53.0 (53cb7b09b 2021-06-17)
```

Tried single-stepping in the debugger; crash appears to be in `proc-macro-2-1.0.27::detection::initialize()`, but the debugger was also acting flaky at this point.
  
[sifive_hifive_unmatched.svd.zip](https://github.com/rust-embedded/svd2rust/files/6780892/sifive_hifive_unmatched.svd.zip)

burrbull, 2021-07-08:
I can't reproduce your issue.
What host system do yo use? How much memory the process takes? Looks like this is not `svd2rust`, but `proc-macro-2` issue.

Try `svd2rust` from master branch and from https://github.com/rust-embedded/svd2rust/compare/svd_ng , please
Does one of them solve your problem?

U007D, 2021-07-09:
Thanks, @burrbull, that is really good to know.

> Try svd2rust from master branch and from https://github.com/rust-embedded/svd2rust/compare/svd_ng
I had tried v0.19.0 from `crates.io` and `master` with the same result.  I tried the `svd_ng` branch today, and it also aborted exactly the same way. :(

> What host system do yo use? 
16GB MacBook Pro, OS X 11.4 (Big Sur).

> How much memory the process takes?
Peak memory usage is 4456 KB (Maximum resident set size according to `gnu-time`).

> Looks like this is not svd2rust, but proc-macro2 issue.
It does look that way.  I'll close this issue as I continue to investigate.  In the meantime, would you be able to post the resulting `.svd` file you generated?  I'd appreciate it!

closed 2021-07-09 by U007D

burrbull, 2021-07-09:
https://drive.google.com/file/d/1Z4lQYrVYaknQuvdK7QhJf5E2s02tXN6C/view?usp=sharing

U007D, 2021-07-09:
@burrbull thank you.

I've also been able to repro your success with the same versions of files and tools under Linux.

I've posted what I _hope_ is a faithful sibling to the `fu-540-pac` crate [here](https://github.com/U007D/fu740-pac).

U007D, 2021-08-01:
Opened an issue in `proc-macro2` repo here: https://github.com/alexcrichton/proc-macro2/issues/292