rust-embedded/svd2rust #547: Reader for bit range [0:0] might generate incorrect code

robamu, 2021-11-09:
I was implementing a HAL implementation for a UART peripheral and came cross odd behaviour:

```rs
            ...
            if status_reader.rdavl().bit_is_set() {
                let data = uart.data.read().bits();
                Ok((data & 0xff) as u8)
            } else {
                Err(nb::Error::WouldBlock)
            }
            ...
```

The `rdavl().bit_is_set()` call always returns `true`,  even when the actual bit in the hardware was not set (I checked in peripheral view and by reading the whole register value as `u32`).

I then checked the `rdavl()` code:

```rs
impl R {
    #[doc = "Bit 0 - Read Data Available"]
    #[inline(always)]
    pub fn rdavl(&self) -> RDAVL_R {
        RDAVL_R::new(self.bits != 0)
    }
    #[doc = "Bit 1 - Read Fifo NOT Full"]
    #[inline(always)]
    pub fn rdnfull(&self) -> RDNFULL_R {
        RDNFULL_R::new(((self.bits >> 1) & 0x01) != 0)
    }
    #[doc = "Bit 2 - RX Busy Receiving"]
    #[inline(always)]
    pub fn rxbusy(&self) -> RXBUSY_R {
        RXBUSY_R::new(((self.bits >> 2) & 0x01) != 0)
    }
```

This is odd. The call does not look correct, it should be `RDAVL_R::new((self.bits & 0x01) != 0)`

I checked the SVD file: 

```xml
          <name>RXSTATUS</name>
          <description>Status Register</description>
          <addressOffset>0x010</addressOffset>
          <access>read-only</access>
          <resetValue>0x0000000</resetValue>
          <fields>
            <field>
              <name>RDAVL</name>
              <description>Read Data Available</description>
              <bitRange>[0:0]</bitRange>
            </field>
            <field>
              <name>RDNFULL</name>
              <description>Read Fifo NOT Full</description>
              <bitRange>[1:1]</bitRange>
            </field>
```

Might this be a bug?

burrbull, 2021-11-09:
Yes, looks like a bug

burrbull, 2021-11-09:
Can you check previous versions to find when causer?

robamu, 2021-11-09:
Yes

robamu, 2021-11-09:
Oh, I actually need another SVD sample file now because the one I always used of course does not work with an older version

burrbull, 2021-11-09:
The problem is after #537

robamu, 2021-11-09:
Okay, thanks. I just tried out the program on the SVD sample file : https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_Example_pg.html . 
The bit mask works there. Maybe it has to do something with the size field missing form the register?

commit 43be074d (2021-11-09): Merge #549

549: fix bug in #537 r=Emilgardis a=burrbull

resolves #547
PR #549 (2021-11-09): fix bug in #537
resolves #547
closed 2021-11-09 by bors[bot]