rust-embedded/svd2rust #557: Clippy complains about generated code

robamu, 2021-12-10:
This is a bit of a luxury problem but maybe something that could be addressed so I don't need to patch a SVD file which is ok in my opinion.

When faced with a register definitions like this:

```xml
        <register>
          <name>WDOGLOAD</name>
          <description>Counter Start Value</description>
          <addressOffset>0x0000</addressOffset>
          <access>read-write</access>
          <resetValue>0xFFFFFFFF</resetValue>
          <fields>
            <field>
              <name>CNT</name>
              <description>Count to load</description>
              <bitRange>[31:0]</bitRange>
            </field>
          </fields>	
        </register>		
 ```
 
 `svd2rust` will generate code which clippy does not like:
 
 ```rs
 error: this operation will always return zero. This is likely not the intended outcome
  --> va416xx/src/watch_dog/wdogload.rs:60:23
   |
60 |         self.w.bits = (self.w.bits & !0xffff_ffff) | (value as u32 & 0xffff_ffff);
   |                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#erasing_op

warning: the operation is ineffective. Consider reducing it to `value as u32`
  --> va416xx/src/watch_dog/wdogload.rs:60:54
   |
60 |         self.w.bits = (self.w.bits & !0xffff_ffff) | (value as u32 & 0xffff_ffff);
   |                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#identity_op

warning: the operation is ineffective. Consider reducing it to `self.bits`
  --> va416xx/src/watch_dog/wdogload.rs:68:20
   |
68 |         CNT_R::new((self.bits & 0xffff_ffff) as u32)
   |                    ^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#identity_op

warning: the operation is ineffective. Consider reducing it to `self.bits`
  --> va416xx/src/watch_dog/wdogvalue.rs:35:20
   |
35 |         CNT_R::new((self.bits & 0xffff_ffff) as u32)
   |                    ^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#identity_op

```

This probably require `svd2rust` to check whether

1. Only one fields exists
2. That field has a width of 32 bits.

In that case, the code can be simplified like clippy suggested

burrbull, 2021-12-10:
@robamu Check #558

robamu, 2021-12-10:
Thanks, this fixes the issue and clippy is happy now :)

closed 2021-12-12 by Emilgardis