rust-embedded/svd2rust #559: svd2rust fails parsing Atmel SAML22 SYNCBUSY register

gtank, 2021-12-20:
I have the SVD for a SAML22N18A ("Atmel SAML22 Series Device Support (1.2.77)" on http://packs.download.atmel.com) and this error:

```
$ svd2rust -i ATSAML22N18A.svd 
[INFO  svd2rust] Parsing device from SVD file
[ERROR svd2rust] Error parsing SVD file
    
    Caused by:
        0: In device `ATSAML22N18A`
        1: In peripheral `GCLK`
        2: In register `SYNCBUSY`
        3: Parsing field `GENCTRL0` at 5375:13
        4: `EnumeratedValue error: Value 2 out of range 0..2
```
Removing the enumerated values allows the parsing to complete without error.

The register is described like this in SVD:
```
        <register>
          <name>SYNCBUSY</name>
          <description>Synchronization Busy</description>
          <addressOffset>0x04</addressOffset>
          <size>32</size>
          <access>read-only</access>
          <fields>
            <field>
              <name>SWRST</name>
              <description>Software Reset Synchroniation Busy bit</description>
              <bitOffset>0</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-only</access>
            </field>
            <field>
              <name>GENCTRL0</name>
              <description>Generic Clock Generator Control 0 Synchronization Busy bits</description>
              <bitOffset>2</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-only</access>
              <enumeratedValues>
                <name>GENCTRL0Select</name>
                <enumeratedValue>
                  <name>GCLK0</name>
                  <description>Generic clock generator 0</description>
                  <value>0x1</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>GCLK1</name>
                  <description>Generic clock generator 1</description>
                  <value>0x2</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>GCLK2</name>
                  <description>Generic clock generator 2</description>
                  <value>0x4</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>GCLK3</name>
                  <description>Generic clock generator 3</description>
                  <value>0x8</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>GCLK4</name>
                  <description>Generic clock generator 4</description>
                  <value>0x10</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>
            [repeats for GENCTRL1-4]
```
and like this in the [datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/60001465A.pdf):
[image]

So it seems like those enumerated values are meant for the entire 5-bit field of `GENCTRL`s, but it's been written in a way that looks like out of range values to svd2rust. I'm new at this, so I'm not sure: is this something svd2rust should handle, or is it just a weird SVD?

Emilgardis, 2021-12-20:
malformed svd, spec does not match what is described in the SVD

Emilgardis, 2021-12-20:
this is what svdconv say

```
*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5428)
  Value 'GCLK2:4' does not fit into field 'GENCTRL1' [3..3:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5428)
  Ignoring EnumeratedValue : 'GCLK2' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5433)
  Value 'GCLK3:8' does not fit into field 'GENCTRL1' [3..3:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5433)
  Ignoring EnumeratedValue : 'GCLK3' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5438)
  Value 'GCLK4:16' does not fit into field 'GENCTRL1' [3..3:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5438)
  Ignoring EnumeratedValue : 'GCLK4' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5458)
  Value 'GCLK1:2' does not fit into field 'GENCTRL2' [4..4:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5458)
  Ignoring EnumeratedValue : 'GCLK1' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5463)
  Value 'GCLK2:4' does not fit into field 'GENCTRL2' [4..4:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5463)
  Ignoring EnumeratedValue : 'GCLK2' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5468)
  Value 'GCLK3:8' does not fit into field 'GENCTRL2' [4..4:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5468)
  Ignoring EnumeratedValue : 'GCLK3' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5473)
  Value 'GCLK4:16' does not fit into field 'GENCTRL2' [4..4:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5473)
  Ignoring EnumeratedValue : 'GCLK4' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5493)
  Value 'GCLK1:2' does not fit into field 'GENCTRL3' [5..5:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5493)
  Ignoring EnumeratedValue : 'GCLK1' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5498)
  Value 'GCLK2:4' does not fit into field 'GENCTRL3' [5..5:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5498)
  Ignoring EnumeratedValue : 'GCLK2' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5503)
  Value 'GCLK3:8' does not fit into field 'GENCTRL3' [5..5:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5503)
  Ignoring EnumeratedValue : 'GCLK3' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5508)
  Value 'GCLK4:16' does not fit into field 'GENCTRL3' [5..5:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5508)
  Ignoring EnumeratedValue : 'GCLK4' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5528)
  Value 'GCLK1:2' does not fit into field 'GENCTRL4' [6..6:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5528)
  Ignoring EnumeratedValue : 'GCLK1' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5533)
  Value 'GCLK2:4' does not fit into field 'GENCTRL4' [6..6:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5533)
  Ignoring EnumeratedValue : 'GCLK2' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5538)
  Value 'GCLK3:8' does not fit into field 'GENCTRL4' [6..6:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5538)
  Ignoring EnumeratedValue : 'GCLK3' (see previous message)

*** ERROR M335: .\svd2rust\ATSAML22N18A.svd (Line 5543)
  Value 'GCLK4:16' does not fit into field 'GENCTRL4' [6..6:1].

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5543)
  Ignoring EnumeratedValue : 'GCLK4' (see previous message)

*** ERROR M338: .\svd2rust\ATSAML22N18A.svd (Line 5410)
  Field 'GENCTRL1' [3..3:1] (read-only) overlaps 'GENCTRL0' [11..2:10] (read-only) (Line 5375)

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5410)
  Ignoring Field : 'GENCTRL1' (see previous message)

*** ERROR M338: .\svd2rust\ATSAML22N18A.svd (Line 5445)
  Field 'GENCTRL2' [4..4:1] (read-only) overlaps 'GENCTRL0' [11..2:10] (read-only) (Line 5375)

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5445)
  Ignoring Field : 'GENCTRL2' (see previous message)

*** ERROR M338: .\svd2rust\ATSAML22N18A.svd (Line 5480)
  Field 'GENCTRL3' [5..5:1] (read-only) overlaps 'GENCTRL0' [11..2:10] (read-only) (Line 5375)

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5480)
  Ignoring Field : 'GENCTRL3' (see previous message)

*** ERROR M338: .\svd2rust\ATSAML22N18A.svd (Line 5515)
  Field 'GENCTRL4' [6..6:1] (read-only) overlaps 'GENCTRL0' [11..2:10] (read-only) (Line 5375)

*** ERROR M211: .\svd2rust\ATSAML22N18A.svd (Line 5515)
  Ignoring Field : 'GENCTRL4' (see previous message)
```

burrbull, 2022-07-19:
Closing as incorrect SVD, not related to svd2rust

closed 2022-07-19 by burrbull