rust-embedded/svd2rust #560: build broken due to new clap_conf version

cfamz, 2021-12-22:
clap_conf released version 0.1.6, which svd2rust will consume, but the build is now broken:
``` 
$ cargo install svd2rust 
...
   Compiling thiserror-impl v1.0.30
   Compiling thiserror v1.0.30
   Compiling svd-parser v0.10.2
   Compiling clap_conf v0.1.6
error[E0107]: this trait takes 0 generic arguments but 1 generic argument was supplied
   --> /home/ANT.AMAZON.COM/chfelder/.cargo/registry/src/github.com-1ecc6299db9ec823/svd2rust-0.19.0/src/main.rs:171:50
    |
171 | fn setup_logging<'a>(getter: &'a impl clap_conf::Getter<'a, String>) {
    |                                                  ^^^^^^     ------ help: remove this generic argument
    |                                                  |
    |                                                  expected 0 generic arguments
    |
note: trait defined here, with 0 generic parameters
   --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/clap_conf-0.1.6/src/lib.rs:78:11
    |
78  | pub trait Getter<'a>: Sized {
    |           ^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0107`.
error: failed to compile `svd2rust v0.19.0`, intermediate artifacts can be found at `/tmp/cargo-install1LgD2d`

Caused by:
  could not compile `svd2rust`
```
Was able to workaround locally by forcing the clap_conf dependency to version 0.1.5.

burrbull, 2021-12-22:
> Was able to workaround locally by forcing the clap_conf dependency to version 0.1.5.
>

Also can be fixed by adding `Out` in 0.1.6:
```rust
fn setup_logging<'a>(getter: &'a impl clap_conf::Getter<'a, Out=String>) {
```

storyfeet, 2021-12-22:
Does your project work now?
I don't really have plans for major changes but if it's buggy I can try to fix.

Emilgardis, 2021-12-23:
> Does your project work now? I don't really have plans for major changes but if it's buggy I can try to fix.

The change from `0.1.5` to `0.1.6` was breaking, so following [cargo semver rules](https://doc.rust-lang.org/cargo/reference/semver.html) (`only changes in the left-most non-zero component are considered incompatible.`) it should have been released as `0.2.0`.

The easiest way to fix it from your side is to yank the 0.1.6 release and re-release it as 0.2.0

We can get the project building again by making the changes needed or specifying the version of `clap_conf` as `=0.1.5`.

It should also possible for people using `cargo install` to specify `--locked`, e.g `cargo install svd2rust --locked`

storyfeet, 2021-12-23:
I was under the impression this didn't apply until version 1.0.0 had already been released

https://semver.org/

Item 4:
``` 
Major version zero (0.y.z) is for initial development. Anything MAY change at any time. The public API SHOULD NOT be considered stable.
```

storyfeet, 2021-12-23:
I've made the switch now. Thanks for your feedback.

closed 2021-12-23 by cfamz