rust-embedded/svd2rust #561: Generate arrya of register when using `%s` in dimElementGroup

duskmoon314, 2022-01-08:
Hi.

In the svd file, I use dimElementGroup to describe a list of registers in this way:

```xml
<register>
  <dim>32</dim>
  <dimIncrement>0x04</dimIncrement>
  <name>LEDC_FIFO_DATA%s</name>
  <description>LEDC FIFO Data Register</description>
  <addressOffset>0x0030</addressOffset>
  <access>read-only</access>
</register>
```

The generated rust code is:

```rust
#[doc = "0x30..0xb0 - LEDC FIFO Data Register"]
pub ledc_fifo_data: [crate::Reg<ledc_fifo_data::LEDC_FIFO_DATA_SPEC>; 32],
```

According to [CMSIS-SVD 1.3.6](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_special.html#dimElementGroup_gr), I believe we use `%s` to describe a list of registers and `[%s]` to describe an array of registers. 

Is this a bug or a feature? I haven't found a description of this behavior in the document.

burrbull, 2022-01-08:
This is not a bug and not a feature. We just decided to produce arrays when it is possible.
You could add PR to change this behavior if you need, but under feature flag.

closed 2022-01-08 by duskmoon314

burrbull, 2022-01-08:
You can try to run `svd2rust` with `--const_generic`. It should create array proxy with can work in more complicated situations.