rust-embedded/svd2rust #578: Array access proposal

burrbull, 2022-02-02:
Current behavior:
  - SVD array has sequential indexes started from 0 and sequential addresses => create Rust array and add it to Cluster/RegisterBlock
  - otherwise create Rust instance for each element of SVD array

Proposal (breaking):
  - For each SVD array with sequential addresses create Rust array. Also create functions with appropriate names to access particular cluster/register in array.
 
So for example instead of:
```rust
struct RegisterBlock {
   ccr1: CCR,
   ccr2: CCR,
   ccr3: CCR,
   ccr4: CCR,
}
```
we will have:
```rust
struct RegisterBlock {
   ccr: [CCR; 4],
}
impl RegisterBlock {
  pub fn ccr1() -> &CCR {
    &self.ccr[0]
  }
  ...
}
```

cc @adamgreig  @therealprof

adamgreig, 2022-02-02:
Looks good to me, I think it's good to have the arrays as much as possible for generic code to access by index, but nice for users to still be able to refer to the normal name `ccr1()` instead of by array `ccr[0]`.

Sh3Rm4n, 2022-02-08:
I really like this proposal. It makes it easier to use this create in HAL implementations, which try to be generic about many details. The current design forces them to use big macro's. 

With this change, the macros would be easier to implement (no more macros with `macro!(ccr1, ccb1, rfw1) ...` and also it enables to use const generics at some places it seems like.

commit 4850a80f (2022-03-15): Merge #579

579: array accessors r=Emilgardis a=burrbull

Closes #578
commit 26f645fa (2022-03-15): Merge #579

579: array accessors r=Emilgardis a=burrbull

Closes #578
commit 7264731e (2022-03-15): Merge #579

579: array accessors r=Emilgardis a=burrbull

Closes #578
PR #579 (2022-03-15): array accessors
Closes #578
closed 2022-03-15 by bors[bot]