rust-embedded/svd2rust #581: svd2rust fails parsing Sipeed GD32VF103 register

kassane, 2022-02-20:
## Reference

[Sipeed Logan nano - svd](https://github.com/sipeed/platform-gd32v/tree/master/misc/svd)

## Output (test)

```bash
[INFO  svd2rust] Parsing device from SVD file
[ERROR svd2rust] Error parsing SVD XML file
    
    Caused by:
        0: In device `GD32VF103`
        1: In peripheral `RTC`
        2: In register `PSCH`
        3: In field `PSC`
        4: Parsing unknown access at 29649:15
        5: unknown access variant 'write' found
```

adamgreig, 2022-02-20:
Thanks for the report! I think this is a problem with the SVD file. On line 29649 the `PSC` field has `<access>write</access>`, but that's not valid SVD, which needs something like `write-only` or `read-write` (see [the spec](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_special.html#elem_access)). Indeed it doesn't validate against the schema:

```
xmllint --schema svd/cmsis-svd.xsd --noout GD32VF103.svd
GD32VF103.svd:18100: element baseAddress: Schemas validity error : Element 'baseAddress': This element is not expected. Expected is one of ( addressBlock, interrupt, registers ).
GD32VF103.svd:29634: element access: Schemas validity error : Element 'access': [facet 'enumeration'] The value 'write' is not an element of the set {'read-only', 'write-only', 'read-write', 'writeOnce', 'read-writeOnce'}.
GD32VF103.svd:29652: element access: Schemas validity error : Element 'access': [facet 'enumeration'] The value 'write' is not an element of the set {'read-only', 'write-only', 'read-write', 'writeOnce', 'read-writeOnce'}.
GD32VF103.svd:29730: element access: Schemas validity error : Element 'access': [facet 'enumeration'] The value 'write' is not an element of the set {'read-only', 'write-only', 'read-write', 'writeOnce', 'read-writeOnce'}.
GD32VF103.svd:29747: element access: Schemas validity error : Element 'access': [facet 'enumeration'] The value 'write' is not an element of the set {'read-only', 'write-only', 'read-write', 'writeOnce', 'read-writeOnce'}.
GD32VF103.svd fails to validate
```

You should patch the SVD file to fix these errors (or report them upstream), and then check with svd2rust again.

burrbull, 2022-02-20:
https://crates.io/search?q=gd32vf103

kassane, 2022-02-20:
I just used the referenced patch from this package and then it worked.
https://github.com/riscv-rust/gd32vf103-pac/blob/master/patches/gd32vf103.yaml

closed 2022-02-20 by burrbull