rust-embedded/svd2rust #582: The ability to obtain offset values for the modifiable bit fields of registers

AndreySmirnov81, 2022-03-14:
To be able to manipulation of peripheral registers through bit-banding, the values of the offset of the modifiable bit fields in the registers are required.

For example, there are functions for bit-banding here: https://docs.rs/stm32f1xx-hal/0.9.0/stm32f1xx_hal/bb/index.html

Implementation variant:
```rust
pub mod perih_name {
   ...
   pub mod reg_name {
      ...
      pub const field_name_OFFSET: u8 = ...;
      ...
   }
   ...
}
```
Example of an implementation variant (stm32f1):
```rust
pub mod rcc {
   ...
   pub mod apb2enr {
      ...
      pub const USART1EN_OFFSET: u8 = 14;
      ...
      impl<'a> USART1EN_W<'a> {
         ...
         pub fn bit(self, value: bool) -> &'a mut W {
            self.w.bits = (self.w.bits & !(0x01 << USART1EN_OFFSET)) | ((value as u32 & 0x01) << USART1EN_OFFSET);
            self.w
         }
      }
      ...
      impl R {
         ...
         pub fn usart1en(&self) -> USART1EN_R {
            USART1EN_R::new(((self.bits >> USART1EN_OFFSET) & 0x01) != 0)
         }
         ...
      }
      ...
   }
   ...
}
```

Usage example:
```rust
unsafe {
   use stm32f1xx_hal::{self as hal, pac};
   hal::bb::set(&(*pac::RCC::PTR).apb2enr, pac::rcc::apb2enr::USART1EN_OFFSET);
}
```

burrbull, 2022-03-14:
I like the idea.

burrbull, 2022-03-14:
But I don't think we need to use it in `bit` or read methods as it give us nothing except compilation deceleration.

Also for bit-banging it is useful only for 1-bitwise writable fields. So the question is do we need constants for all fields or only for such ones.

RamDav, 2023-04-01:
I have the same problem an would need this. In c i usually liked to build my register initialization values from macros which used the values provided by the CMSIS peripherals files. and use this to setup a register with one compile time calculated value.