rust-embedded/svd2rust #584: safe manipulation of multi-bit values

thepperson, 2022-03-22:
It looks like the only to modify a multi-bit field is with the bits() function, which is marked as unsafe. What is the reasoning behind this being an unsafe function? Is there a safe equivalent for setting a multi-bit field of a register in a peripheral?

Emilgardis, 2022-03-22:
if the svd the PAC is generated from does not specify how those bits are valid there's no way to ensure usage is "valid". Those bits would be accessed through enumerated values

I don't think there is a way in svds to specify that "yes, all bit patterns are valid"

adamgreig, 2022-03-22:
For a particular field, svd2rust marks `bits()` as `safe` if either there are enumerated values that cover all possible bit patterns, or the field has a `writeConstraint` attribute in the SVD that specifies that all possible values for the field width are permitted.

In general you should hopefully use the methods generated by the enumerated values for many multi-bit fields. Whether it should be "unsafe" to write an invalid value to a field is a different question but at the moment that's how svd2rust works.

thepperson, 2022-03-22:
Ah, that makes a lot of sense. Thank you for the feedback.

closed 2022-03-22 by thepperson