rust-embedded/svd2rust #593: Links in the generated package description are not being rendered in documentation

jessebraham, 2022-04-22:
I noticed this after releasing then newest versions of our PACs today. For examples, see:  
https://docs.rs/esp32/0.12.0/esp32/  
https://docs.rs/esp32c3/0.4.0/esp32c3/  
https://docs.rs/esp32s2/0.2.0/esp32s2/  
https://docs.rs/esp32s3/0.2.0/esp32s3/  
https://docs.rs/esp8266/0.6.0/esp8266/

You can see the `[here]`, `[next]`, and `[repository]` placeholders still present, with their associated URLs being rendered as text in the final paragraph. Just to be thorough I did test this locally as well with the same results.

Emilgardis, 2022-04-22:
This is either a rustdoc bug or quote bug.

what's the rustfmt config like?

Emilgardis, 2022-04-22:
I was wrong, this is the issue

https://github.com/esp-rs/esp-pacs/blob/909f0b0c51d8242ac87e23bd00d7e1bd999c969b/xtask/src/main.rs#L124

jessebraham, 2022-04-22:
Ahh jeez silly mistake, thanks for finding that. Sorry for the false alarm!

closed 2022-04-22 by jessebraham

Emilgardis, 2022-04-22:
simple fix for it would be to skip replacing the first few lines, better alternative would be to run `rustfmt`, you can invoke it on the svd2rust output with something like

```rust
use std::process::Command;
let echo = Command::new(format!(r#"echo {}"#, "items")).spawn().unwrap().stdout.unwrap();
let res = Command::new("rustfmt").args(["--edition", "2018"]).stdin(echo).output().unwrap();
let data = if res.status.success() {
    res.stdout
} else {
    panic!("rustfmt failed");
};
```